mne-tools/mne-python #11630: Importing bad blocks description from Raw neuroscan cnt files.

withmywoessner, 2023-04-14:
### Describe the new feature or enhancement

Hello! My lab has a lot of old data that was processed using neuroscan, and I am trying to import some of our raw cnt files into mne. I have run into a problem with importing event files as neuroscan uses the same stim code to mark the end and the beginning of bad blocks (data spans): 
[image]
As far as I can tell, mne only imports the event code and the onset of these events (not the duration). However, I was looking through an old manual on neuroscan and it references another parameter that describes whether you should accept or reject a block:
[image]
I was wondering if this could be implemented by appending '_bad' to the end of the blocks that are meant to be rejected and manually computing the duration of the event by finding the difference in onsets of the event codes.

### Describe your proposed implementation

From cnt.py:
[image]
Make a new variable:
`accept = np.array([str(e.KeyPad_Accept) for e in my_events])
If it contains 'xc0' append '_bad' to the event code (description of event in mne)
Manually compute durations (currently all durations are 0) by finding the difference between codes.

### Describe possible alternatives

I cannot think of any.

agramfort, 2023-04-15:
what I would recommend here is to send us a pull request with code snippet
and a file so can test too and see the diff in the code.

withmywoessner, 2023-04-15:
> what I would recommend here is to send us a pull request with code snippet and a file so can test too and see the diff in the code.
> […](#)

I am new to using github. Do I need to make commits of my proposed changes?

withmywoessner, 2023-04-15:
I'm sorry. I am very new to github and software development. Do I need to
make a separate branch and commit to show you this code snippet?

withmywoessner, 2023-04-16:
What's the best way for me to send a file in my push? The data is 38MB so I cannot upload it as a zip. Thank you!

agramfort, 2023-04-16:
we put our testing data in https://github.com/mne-tools/mne-testing-data

for now you can just share a link to the file ? We can help to put the file
in https://github.com/mne-tools/mne-testing-data
the smaller is the file the better.

> Message ID: ***@***.***>
>

PR #11631 (2023-05-01): ENH: Fix CNT annotations to include bad spans
#### Reference issue
Fixes #11630.

#### What does this implement/fix?
When importing raw neuroscan cnt files important information regarding bad spans is not included. This is because Neuroscan marks the start and end of bad blocks with event code 0. Currently, mne.io.read_raw_cnt only reads the description (StimType ), onset (Offset), and duration (Latency) of the events. However, there is another parameter KeyPad_Accept that includes whether the event code marks the beginning of a bad or good span (or neither). 
[image]

In my proposed changes I [... cut]
closed 2023-05-01 by larsoner