mne-tools/mne-python #11636: Remove Discord link from documentation navigation

hoechenberger, 2023-04-19:
### Proposed documentation enhancement

Currently we prominently link to Discord from our documentation's top navigation.

Considering that we actually don't **want** to direct users to Discord, except for the office hours and sprints – which are supposed to be announced specially anyway – I'd like to propose the removal of the Discord link. I mean, what currently happens is that users come to our website looking for help, find Discord, follow that link, only to be told that they're not supposed to ask questions there (**if** they do read that memo). It doesn't make sense in my opinion.

cbrnr, 2023-04-19:
Good idea, you are absolutely right!

drammock, 2023-04-19:
-1 from me.  Hiding our discord won't fix the problem, and it makes our office hours less discoverable.  This week's dev meeting will include discussion of some feedback we got from sprint attendees, some of which regards how we use discord.  I suspect some changes will happen soon; watch for the meeting minutes if you can't make the meeting itself.

cbrnr, 2023-04-19:
> Hiding our discord won't fix the problem, and it makes our office hours less discoverable.

I think it will improve things. If we don't link to Dicord, then people won't find it and therefore won't come and ask questions there. Do you not agree?

Regarding making the office hourse less discoverable, yes, that's a valid concern, but we announce it in Discourse anyway. So why would a link to Discord at the top of our website impact that?

Anyway, looking forward to what's being discussed at the dev meeting.

drammock, 2023-04-19:
> Regarding making the office hourse less discoverable, yes, that's a valid concern, but we announce it in Discourse anyway.

We don't know whether (and shouldn't assume that) every user who clicks the discord link will also browse the forum long enough / frequently enough / at the right time to discover the office hours announcements.

> So why would a link to Discord at the top of our website impact that?

A new user joining our Discord server will see the following:

- rules
- announcements
- jobs
- bids
- public voice channels
    - temporal lobe, etc
- office hours
    - office hours readme   <---- THIS
    - office hours
- archived channels

> I think it will improve things. If we don't link to Dicord, then people won't find it and therefore won't come and ask questions there. Do you not agree?

I agree that it might reduce unwanted support requests on Discord. But I'd rather make our Discord space *more useful* (while still respecting devs' time/preferences) rather than *less findable*

cbrnr, 2023-04-20:
Just to make sure, do we want users to use Discord for anything else than office hours? It seems that a lot of stuff is duplicated from Discourse (announcements, jobs), and I'd prefer to have only one place for public discussion/support (Discourse and not Discord).

drammock, 2023-04-20:
> do we want users to use Discord for anything else than office hours?

The community seems to want to use it for things other than office hours. I think we should be responsive to that *in some way*. That doesn't necessarily mean we "cave in" and start answering support questions there, but I don't think it's working to just ignore the problem.
(Aside: in addition to office hours we already use it for our sprints and dev meetings, and [braindecode](https://braindecode.org/stable/index.html) is going to do a sprint in our Discord this summer.)

> It seems that a lot of stuff is duplicated from Discourse (announcements, jobs)

Fair point. it would indeed be better if Discord jobs channel and Discourse jobs topic were in sync.  I'm hesitant to say one of them should just go away though.  Maybe the same applies to announcements though I'm less certain of that.

britta-wstnr, 2023-04-21:
My 2 cents: I think the webpage should link Discord as well as our office hours rules (like "usually at this time and this place, but check Discourse for up-to-date information").
Just last week I was searching for a link to send to a collaborator and could not find anything - the most up-to-date one I could find was an announcement from February 16th (I know we have a calendar but not everybody wants to clutter their own calendar with the office hours calendar to just learn about when it happens _generally_).

hoechenberger, 2023-04-21:
> My 2 cents: I think the webpage should link Discord as well as our office hours rules (like "usually at this time and this place, but check Discourse for up-to-date information").

I like the idea of linking to a new page that first says what Discord is all about before taking you there, instead of linking to Discord directly as we do now

larsoner, 2023-10-02:
@drammock and I went through and cleaned up discord quite a bit. We:

1. removed the redundancy of announcements and jobs as suggested by @cbrnr 
2. moved office-hours up to the top along with a `# readme` that almost immediately mentions that the channel is for interactive office hours discussions
3. added a join message that suggests only the office-hours channel for interactive sessions, and suggests using Discourse if it's outside office hours

With that do you think we can close @hoechenberger ?

hoechenberger, 2023-10-02:
@larsoner Yes, ok to close. Thanks

closed 2023-10-02 by hoechenberger