mne-tools/mne-python #11642: MAINT: Release 1.4

larsoner, 2023-04-21:
We should release 1.4 in a week or two if possible. I've gone through and bumped milestones on a bunch of issues/PRs to [1.5](https://github.com/mne-tools/mne-python/milestone/26) and left what I think is hopefully doable for 1.4 [here](https://github.com/mne-tools/mne-python/milestone/25).

@mne-tools/mne-python-steering-committee if you want, feel free to take a look at the open issues and PRs early next week and mark things you think can realistically get in within a short time frame for 1.4!

larsoner, 2023-05-01:
Looking at https://github.com/mne-tools/mne-python/milestone/25 I think the only blocker that isn't a PR by me is the documentation one https://github.com/mne-tools/mne-python/issues/6794 @drammock . I'm thinking we should try to release by the end of the week. Let me know if there's anything you want me to wait for in #6794, otherwise I'll get the release process going once my PRs are in.

drammock, 2023-05-01:
the old docs issue (#6794) doesn't have any blockers IMO. But getting our `conf.py` and CSS in sync with recent changes to the theme is on my list for this week (resuming work on it today in fact).

larsoner, 2023-05-02:
FYI https://github.com/numba/numba/releases/tag/0.57.0 just landed so it's possible that we might be able to use Python 3.11 in our installers if the conda-forge machinery works without issue in the next few days!

hoechenberger, 2023-05-03:
@larsoner I reached out to the team via gitter, let's see if somebody gets back to us

hoechenberger, 2023-05-03:
It got merged!

larsoner, 2023-05-04:
@drammock ready for release from your end? If so I'll get the process going tomorrow

drammock, 2023-05-04:
I think we're good to go. I thought there were some lingering deprecations to remove (related to topomap param changes IIRC) but now I'm not finding them.  One possibile breaking change that might be good to sneak in is the TODO item here:

https://github.com/mne-tools/mne-python/blob/dba7df1914f020f8905c7be4ae5a0ce823e4e2eb/mne/utils/spectrum.py#L66

If you have time to do that, great.  If not I can do tomorrow morning if you're willing to not start the release first thing.  Or it can wait until next version.

larsoner, 2023-05-04:
Feel free to do it tomorrow when you get a chance

larsoner, 2023-05-05:
Don't want to start the release process this late on a Friday because it risks having to fix stuff over the weekend. I'll start it Monday instead