mne-tools/mne-python #11645: tfr_array_morlet issues with wavelet

RainyMeadows, 2023-04-23:
### Description of the problem

.

### Steps to reproduce

```Python
import numpy as np
from mne.time_frequency import tfr_array_morlet

freqs=np.arange(1, 30, 1)
power = tfr_array_morlet(Input_data, 1000, freqs, n_cycles=6.0, output='avg_power') # Input data is a 70 * 128 * 6000 array
```

### Expected results

An array of shape (n_chans, n_freqs, n_times)

### Actual results

ValueError: At least one of the wavelets is longer than the signal. Use a longer signal or shorter wavelets.

### Additional information

Platform:         Windows-10-10.0.22621-SP0
Python:           3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\user\AppData\Local\Programs\Python\Python310\python.exe
CPU:              Intel64 Family 6 Model 167 Stepping 1, GenuineIntel: 12 cores
Memory:           15.7 GB

mne:              1.3.1
numpy:            1.24.2 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.10.1

agramfort, 2023-04-23:
your minimum frequency is too low. 6 cycles at 1 Hz is longer than your
signal given the way the Morlet wavelet is clipped temporally.

Message ID: ***@***.***>
>

RainyMeadows, 2023-04-23:
What is the minimum signal length - some handy calculator? I was [assuming](https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2018-December/005398.html) that at lowest (1 Hz), 7 cycles equate to 7s of signal.

agramfort, 2023-04-24:
look at the code. I don't know on the top of my head how the clipping is
done.

> Message ID: ***@***.***>
>

RainyMeadows, 2023-04-24:
See my (edited) comment; l I did take a look!

agramfort, 2023-04-25:
@RainyMeadows would you be up for submitting a PR to update our doc? that would be much appreciated. thanks

PR #12073 (2023-10-06): DOC: Morlet wavelet length in tfr_morlet
Closes #11645.

This PR adds clarification of the Morlet wavelet temporal window length used with `mne.time_frequency.tfr_array_morlet` and `mne.time_frequency.tfr_morlet`.
closed 2023-10-06 by drammock