mne-tools/mne-python #11647: bad pre-commit install spec

drammock, 2023-04-23:
### Description of the problem

in a fresh env (where I did `mamba install --only-deps mne` to initialize it), installing the pre-commit hooks fails because it expects python 3.11 but our condaforge recipe installs python 3.10

### Steps to reproduce

```Python
mamba install --only-deps mne
mamba install pre-commit
cd /path/to/mne-python
pre-commit install --install-hooks
```
```

### Expected results

pre-commit hooks get installed

### Actual results

```
$ pre-commit install --install-hooks
pre-commit installed at .git/hooks/pre-commit
[INFO] Installing environment for https://github.com/charliermarsh/ruff-pre-commit.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
An unexpected error has occurred: CalledProcessError: command: ('/opt/mambaforge/envs/mnedev/bin/python3.10', '-mvirtualenv', '/home/drmccloy/.cache/pre-commit/repo90m6g1k7/py_env-python3.11', '-p', 'python3.11')
return code: 1
stdout:
    RuntimeError: failed to find interpreter for Builtin discover of python_spec='python3.11'
stderr: (none)
Check the log at /home/drmccloy/.cache/pre-commit/pre-commit.log
```

### Additional information

```
Platform             Linux-5.15.0-58-generic-x86_64-with-glibc2.35
Python               3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:08:06) [GCC 11.3.0]
Executable           /opt/mambaforge/envs/mnedev/bin/python3.10
CPU                  x86_64 (16 cores)
Memory               125.7 GB

Core
├☑ mne               1.4.0.dev127+g9145025c4.d20230423
├☑ numpy             1.23.5 (unknown linalg bindings)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=QtAgg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.56.4
├☑ nibabel           5.1.0
├☑ nilearn           0.10.0
├☑ dipy              1.6.0
├☑ openmeeg          2.4.7
├☑ pandas            2.0.0
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.38.5 (OpenGL 4.5.0 NVIDIA 470.161.03 via NVIDIA GeForce RTX 2060/PCIe/SSE2)
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.4.dev0
└☐ unavailable       ipympl

Ecosystem (optional)
├☑ mne-bids          0.12.dev0
├☑ mne-connectivity  0.6.0dev0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline
```

cbrnr, 2023-04-23:
If we want to stick with 3.10, we need to add tomli as a dependency I think.

drammock, 2023-04-23:
> If we want to stick with 3.10, we need to add tomli as a dependency I think.

that seems like a pretty light dependency to add, no objection from me

PR #11648 (2023-04-25): use py3.10 in precommit config
closes #11647 

draft for now until it's clear where the right place is to put `tomli` (I put in `requirements_testing.txt` but that seems not ideal).

Also I expect the CI configs will need updating too?
closed 2023-04-25 by larsoner