mne-tools/mne-python #11651: Allow for computation of TFR using Morlet wavelets but with fwhm as an alternative parameter to wave-number

RainyMeadows, 2023-04-24:
### Describe the new feature or enhancement

It is nice to see `mne.time_frequency.fwhm` but it is a tad useless (except for final reporting), unless we can use `fwhm` values to construct the wavelets. For an instructive case, where this comes handy, see **Fig. 5 a (var)** in Cohen (2019).

### Describe your proposed implementation

As a new capability to `mne.time_frequency.morlet` and like methods.

### Describe possible alternatives

None

drammock, 2023-04-25:
IIRC there was some discussion of this back when https://github.com/mne-tools/mne-python/pull/11353 was being worked on.  @larsoner @britta-wstnr WDYT?  Can we support constructing wavelets based on their FWHM, and if so, what is a good API to do that?

britta-wstnr, 2023-04-26:
Back then we added the formula to compute the FWHM to the docstring (e.g. of `mne.time_frequency.morlet`). Using this formula, also the adequate number of cycles for a desired FWHM can be computed and then passed to the function.

That being said, we could support this via the functions as well - I think we discussed it back in December at some point but I don't remember the pro's and con's anymore, @larsoner ?
If we want to add it, I guess a new parameter (maybe `wav_fwhm` or so)  that is mutually exclusive with `n_cycles` would be needed.

larsoner, 2023-04-26:
I think the idea was to avoid adding a new `kwarg` by having this convenience conversion function. But we could have mutually exclusive `kwargs` instead if it really helps

drammock, 2023-04-26:
> I think the idea was to avoid adding a new `kwarg` by having this convenience conversion function

The function goes the wrong direction though: it takes in `freq` and `n_cycles` and spits out `fwhm`.

I can open a PR to add this, I think it's worthwhile.

PR #11658 (2023-04-28): clarify relationships among FWHM, sigma, and n_cycles in Morlet function docstrings
closes #11651
closed 2023-04-28 by larsoner