mne-tools/mne-python #11653: ICA plot_components on provided axes fails

mscheltienne, 2023-04-24:
### Description of the problem

```
from matplotlib import pyplot as plt
from mne.datasets import sample
from mne.io import read_raw_fif
from mne.preprocessing import ICA

# create a toy dataset
fname = sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = read_raw_fif(fname, preload=False)
raw.pick_types(eeg=True)
raw.crop(0, 10)
raw.load_data()
raw.filter(1.0, 40.0)
ica = ICA(n_components=5, method="picard")
ica.fit(raw)

# create figure
f, ax = plt.subplots(2, 2)
ica.plot_components(inst=raw, picks=[1, 2, 3, 4], axes=ax)
```

The code snippet above raises:

```
AttributeError: 'numpy.ndarray' object has no attribute 'set_title'
```

If you flatten the provided `ax`, it fails with:

```
UnboundLocalError: local variable 'fig' referenced before assignment
```

Working on it atm.

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

hoechenberger, 2023-04-24:
Looking at the docstring, this is actually the expected behavior and hence not a bug:

> axesinstance of [Axes](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.html#matplotlib.axes.Axes) | [list](https://docs.python.org/3/library/stdtypes.html#list) of [Axes](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.html#matplotlib.axes.Axes) | [None](https://docs.python.org/3/library/constants.html#None)
The axes to plot to. If None, a new [Figure](https://matplotlib.org/stable/api/figure_api.html#matplotlib.figure.Figure) will be created with the correct number of axes. If [Axes](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.Axes.html#matplotlib.axes.Axes) are provided (either as a single instance or a [list](https://docs.python.org/3/library/stdtypes.html#list) of axes), the number of axes provided must match the number of times provided (unless times is None).Default is None.

Since you pass a multi-dimensional array and not a list-like object, the method fails.

The only bug I can see here is that the error message that is produced is not super helpful – we're obviously failing too late.

I do appreciate your efforts in #11640 though, but you're actually adding a new functionality there (supporting `NDarray input), so it's not a bug fix and would also require you to rework the docstring a bit.

mscheltienne, 2023-04-24:
The `UnboundLocalError` is definitely a bug, it's also present if you pass a list of axes. But you are right that at the moment the docstring says `List of Axes`. The same docstring also mentions `times` for ICA topographic maps.. 😅 

My reasoning would be that an `axes` argument (or its plural form) should support any output from a `matplotlib` function that creates axes, as long as the correct number of subplots is provided. 

Sure, you can create them one by one, e.g.

```
from matplotlib import pyplot as plt

f = plt.Figure()
axes = list()
for k in range(4):
    axes.append(f.add_subplot(2, 2, k + 1))
```

But you can also create them all at once with `plt.subplots` which returns always a 2D array of axes, except if `squeeze=True` (the default). IMO, output from `plt.subplots` should be supported out of the box by an `axes` argument which can take multiple subplots. 

It's difficult to make one `axes` docstring for all MNE-function, but for `ica.plot_components`, the docstring for `axes` would become:

```
axes : Axes | array of Axes | None
    Either a single Axes or an iterable of Axes if more than one subplot
    is needed. The number of subplots must match the number of 
    selected components. If None, a new figure will be created
    The subplot(s) to plot to. The number of subplots must match the
    number of selected components. If None, new figures will be created
    with the number of subplots per figure controlled by ``nrows`` and ``ncols``
```

It's not complete as the types at the top don't mention iterable to avoid lengthening the line to:

```
axes : Axes | array of Axes | iterable of Axes | None
```

And it does not mention the 1D / 2D `ndarray`, but I think that would be adding too much detail.

hoechenberger, 2023-04-24:
@larsoner 
> To me if `list of axes` works then `1D ndarray of axes` should also work,

But @mscheltienne passed a 2d array of `axes` in the example

@mscheltienne 
> MO, output from `plt.subplots` should be supported out of the box by an `axes` argument which can take multiple subplots.

I generally agree, but how would you determine the order in which to fill the axess (see what I did :D)? Is left -> right from top to bottom what users would expect? (I'd say: yes, but I'm not sure if everybody would agree?)

larsoner, 2023-04-24:
> But @mscheltienne passed a 2d array of axes in the example

Ahh I missed that!

> I generally agree, but how would you determine the order in which to fill the axess (see what I did :D)? Is left -> right from top to bottom what users would expect? (I'd say: yes, but I'm not sure if everybody would agree?)

I think it's pretty clear by the default C-ordering of NumPy arrays that we will just `.ravel()` / `.flatten()` the array, which means it will go across the first row left to right, then the second row left to right, etc. If people want something else they can pass `axes.T` or explicitly order them.

PR #11654 (2023-04-25): Fig bug with ica.plot_components when axes is provided
Closes #11653