mne-tools/mne-python #11659: BUG: MNE `concatenate_raws` duplicates loaded data if run in a loop

GABowers, 2023-04-27:
### Description of the problem

The `Raw` object generated from the `concatenate_raws` method duplicates data if run on the same **sequence** of `Raw` objects more than once.

I first noticed this on my data which is using the .fif file format, but as can be seen below it occurs for edf files as well.

### Steps to reproduce

```Python
import mne, mne.datasets.eegbci

if __name__ == "__main__":
    mne.set_log_level(verbose='ERROR')
    raw_fnames = mne.datasets.eegbci.load_data(1,  [*range(1, 15)])
    raws = [mne.io.read_raw_edf(f, preload=True) for f in raw_fnames]
    for i in range(5):
        sub = [x for x in raws[:5]]
        # if the line above is instead "raws[i:i+5]]" the error does not occur
        main = mne.concatenate_raws(sub, verbose='INFO')
        if len(sub) != len(main.filenames):
            raise Exception("Too many files in Raw object!")
        print('{} files loaded'.format(len(main.filenames)))
        del(main)
        del(sub)
```

### Expected results

The concatenated `Raw` object should only include files specified the most recent collection of `Raw` objects

### Actual results

The concatenated `Raw` object includes duplicate data.

### Additional information

Platform:         Windows-10-10.0.19044-SP0
Python:           3.11.3 (tags/v3.11.3:f3909b8, Apr  4 2023, 23:49:59) [MSC v.1934 64 bit (AMD64)]
Executable:       C:\Users\~\AppData\Local\Programs\Python\Python311\python.exe
CPU:              AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD: 16 cores
Memory:           63.9 GB

mne:              1.3.1
numpy:            1.24.2 {OpenBLAS 0.3.21 with 16 threads}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=TkAgg}

sklearn:          1.2.2
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           2.0.0
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.7.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2023-04-27:
see the note section in
https://mne.tools/stable/generated/mne.concatenate_raws.html

can it explain what you see?

> Message ID: ***@***.***>
>

GABowers, 2023-04-27:
> see the note section in https://mne.tools/stable/generated/mne.concatenate_raws.html can it explain what you see?
> […](#)
> Message ID: ***@***.***>

Ah yes, I think it does. Thanks for pointing that out.

closed 2023-04-27 by GABowers