mne-tools/mne-python #11660: mne.io.Raw.filter Error with NIRS data

Unknolwn, 2023-04-27:
### Description of the problem

Hello, I’m working with NIRS data have trouble updating my code to MNE latest version because of the following MNE filtering function stops working when I do. [mne.io.Raw — MNE 1.2.2 documentation](https://mne.tools/1.2/generated/mne.io.Raw.html#mne.io.Raw.filter)

    MNE version: 1.0.3
    MNE-BIDS version : 0.10
    MNE-NIRS version : 0.5.0
    operating system: Windows 10

### Steps to reproduce

```Python
Here is an example of a working code with MNE 1.0.3 but with MNE 1.1 or 1.3 I have a warning with same function : 
import mne
import mne_bids
import mne_nirs
#steps to import data
BidsPath="C:/datafolder"
dataset = mne_bids.BIDSPath(root=BidsPath, task=mne_bids.get_entity_vals(BidsPath, 'task')[0], datatype="nirs", suffix="nirs", extension=".snirf")
subjects = mne_bids.get_entity_vals(BidsPath, 'subject')
sessions = mne_bids.get_entity_vals(BidsPath,'session')
bids_path=dataset.update(subject=subjects[0])
bids_path=dataset.update(session=sessions[0])
raw_data=mne_bids.read_raw_bids(bids_path=bids_path, extra_params=None, verbose=False)
#convert data
OD=mne.preprocessing.nirs.optical_density(raw_data)
Conc=mne.preprocessing.nirs.beer_lambert_law(OD,ppf=6)
#filter data
Filtered=Conc.copy().filter(0.05,0.1, l_trans_bandwidth=0.01, h_trans_bandwidth=0.01, verbose=False)
```

### Link to data

You can find NIRS dataset here:  https://osf.io/b4wck/

### Expected results

Filtered data with only this warning:
mne\filter.py:312: DeprecationWarning: Keyword argument 'nyq' is deprecated in favour of 'fs' and will be removed in SciPy 1.12.0.
  this_h = firwin(this_N, (prev_freq + this_freq) / 2.,

### Actual results

Result with MNE 1.3:

File "<stdin>", line 1, in <module>
  File "<decorator-gen-612>", line 10, in read_raw_bids
  File "C:\environment\lib\site-packages\mne_bids\read.py", line 713, in read_raw_bids
    raw = _handle_channels_reading(channels_fname, raw)
 File "C:\environment\lib\site-packages\mne_bids\read.py", line 552, in _handle_channels_reading
    raw.rename_channels({raw_ch_name: bids_ch_name})
  File "<decorator-gen-43>", line 12, in rename_channels
  File "C:\environment\lib\site-packages\mne\channels\channels.py", line 414, in rename_channels
    rename_channels(self.info, mapping, allow_duplicates)
  File "<decorator-gen-51>", line 12, in rename_channels
  File "C:\environement\lib\site-packages\mne\channels\channels.py", line 1112, in rename_channels
    raise ValueError('New channel names are not unique, renaming failed')
ValueError: New channel names are not unique, renaming failed

### Additional information

Platform:         Windows-10-10.0.19044-SP0
mne:              1.0.3
numpy:            1.22.3 {}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=TkAgg}

sklearn:          1.2.2
numba:            Not found
nibabel:          5.1.0
nilearn:          0.10.0
dipy:             Not found
cupy:             Not found
pandas:           2.0.0
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
PyQt5:            Not found
ipympl:           Not found
pooch:            v1.7.0

mne_bids:         0.10
mne_nirs:         0.5.0

larsoner, 2023-04-27:
> the following MNE filtering function stops working when I do

To me it looks like it's the *reading* function that fails much earlier than the *filtering* function. Can you confirm that the failure is actually up at the `read_raw_bids` step?

hoechenberger, 2023-04-28:
> To me it looks like it's the reading function that fails

Yes, according to the traceback, this issue has nothing to do with filtering. The issue should be transferred over to mne-tools/mne-bids

Unknolwn, 2023-04-28:
Oh, okay, I'm going to do that

Unknolwn, 2023-04-28:
I found this issue (https://github.com/mne-tools/mne-bids/issues/1117) on MNE_BIDS that looks like mine but the conclusion seems to lead back to MNE-python.

larsoner, 2023-04-28:
In https://github.com/mne-tools/mne-bids/issues/1117 I think the conclusion was that MNE-BIDS needs to take care of ordering differences between the data and `channels.tsv`. Not sure if it's the same problem, will have to test...