mne-tools/mne-python #11671: error in bandwidth in mne.time_frequency.psd_array_multitaper

chapochn, 2023-05-02:
### Proposed documentation enhancement

In the function mne.time_frequency.psd_array_multitaper, the documentation for `bandwidth` is:

> bandwidth: Half-bandwidth of the multi-taper window function in Hz. For a given frequency, frequencies at ± half-bandwidth are smoothed together. The default value is a half-bandwidth of 4.

However, in the code I see it is calling:
`_compute_mt_params(n_times, sfreq, bandwidth, low_bias, adaptive)`

And inside `_compute_mt_params`:
`half_nbw = float(bandwidth) * n_times / (2. * sfreq)`

meaning that it is interpreting bandwidth as the full bandwidth, not as the half-bandwidth as the documentation of `psd_array_multitaper` is saying. My tests confirm that `psd_array_multitaper` is using the full bandwidth. One should either change the code of the documentation. Thoughts?

larsoner, 2023-05-02:
I think this has been fixed in https://github.com/mne-tools/mne-python/pull/11479 and should be released in 1.4 this week. Feel free to reopen if I'm mistaken!

https://mne.tools/dev/generated/mne.time_frequency.psd_array_multitaper.html

chapochn, 2023-05-02:
ok great