mne-tools/mne-python #11684: Annotations do not support Unicode characters

hoechenberger, 2023-05-09:
### Description of the problem

Raw data (and possibly epochs, which I haven't tested) with Annotations that contain Unicode descriptions cannot be saved.

This was initially reported at https://mne.discourse.group/t/saving-filtered-data-and-epochs/6848/3

### Steps to reproduce

```Python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname, preload=True)
raw.crop(tmax=60)

annots = mne.Annotations(onset=0, duration=1, description='🙃')
raw.set_annotations(annots)

raw.save('/tmp/foo-raw.fif', overwrite=True)
```

### Expected results

File is saved

### Actual results

```python
Writing /tmp/foo-raw.fif
Traceback (most recent call last):
  File "/private/tmp/mwe.py", line 13, in <module>
    raw.save('/tmp/foo-raw.fif', overwrite=True)
  File "<decorator-gen-242>", line 12, in save
  File "/Users/hoechenberger/Development/mne-python/mne/io/base.py", line 1691, in save
    _write_raw(
  File "/Users/hoechenberger/Development/mne-python/mne/io/base.py", line 2604, in _write_raw
    cals = _start_writing_raw(
  File "/Users/hoechenberger/Development/mne-python/mne/io/base.py", line 2872, in _start_writing_raw
    _write_annotations(fid, annotations)
  File "/Users/hoechenberger/Development/mne-python/mne/annotations.py", line 1084, in _write_annotations
    write_name_list_sanitized(
  File "/Users/hoechenberger/Development/mne-python/mne/io/write.py", line 149, in write_name_list_sanitized
    write_string(fid, kind, _safe_name_list(lst, "write", name))
  File "/Users/hoechenberger/Development/mne-python/mne/io/write.py", line 130, in write_string
    str_data = str(data).encode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\U0001f643' in position 0: ordinal not in range(256)
```

### Additional information

Since Unicode Annotations can be set, we should be able to write them. Hence I'm reporting this as a bug and not a feature request.

Proposal by @cbrnr, @drammock, and me:

- prevent using non-latin1 characters in Annotations (@hoechenberger) – but this is actually not acceptable in 2023 (@drammock, and I fully agree with that)
- amend FIFF definition to allow Unicode character storage (@cbrnr, @drammock)
- provide HDF5 export for al affected file types (@drammock)
- Maybe we can sneak in the Unicode via an ASCII encoding, think base64, but this sounds rather ugly (@hoechenberger)

larsoner, 2023-10-02:
... also `latin-1` is not a subset of `utf-8` so backward compat here would be nearly impossible it seems -- writing utf-8 then reading with latin-1 software (MEGIN or old MNE or MNE-MATLAB, etc.) can get garbled stuff, and vice-versa (e.g., https://github.com/mne-tools/fiff-constants/commit/82aaf647c91fd2565f9a868065b1e9003c4279e4 rendering funny on GitHub by latin-1 encoding the superscript 3). So a compatible path forward is not trivial :(

hoechenberger, 2023-10-02:
What about this workaround:

> * provide HDF5 export for all affected file types (@drammock)

We've resorted to using HDF5 to overcome FIFF limitations in several other places, too.

When the user sets non-latin-1 strings, we can emit a warning that from now on, only HDF5 is allowed for an I/O round-trip, and raise an exception when they save as FIFF. Or raise another warning in that case, and force-convert to latin-1 as good as we can.

larsoner, 2023-10-02:
To me adding support for writing annot to HDF5 isn't really desirable because you still can't do `raw.set_annotations(annot); raw.save(...)` and have it work. So it's weird to have this partial support. And having `raw.save('...-raw.h5')` doesn't make sense because it's not portable to other software, which I think is important for raw data saving. (HDF5 in principle is readable everywhere but it's not a known/defined format for M/EEG data import in any other package.)

PR #12080 (2023-10-06): add unicode roundtrip for FIF
closes #11684
closed 2023-10-06 by larsoner