mne-tools/mne-python #11693: ICA plot_components opens figure one by one (and might hang depending on the console)

mscheltienne, 2023-05-12:
### Description of the problem

A video will be clearer. Same behavior in a regular python console and in an IPython console.

[Screencast from 05-12-2023 04:45:47 PM.webm](https://github.com/mne-tools/mne-python/assets/73893616/fdd3f042-9e24-4ff2-90c6-c59792596ef0)

The figure are shown one by one after you close the old one. At least in a python or IPython console, it works. In spyder, it hangs the console after you close the first figure :disappointed: 

Tested both in a working environment and in a fresh environment with only `pip install ipython mne` installed (MNE 1.4).

### Steps to reproduce

```Python
from mne.datasets import sample
from mne.io import read_raw_fif
from mne.preprocessing import ICA

directory = sample.data_path() / "MEG" / "sample"
raw = read_raw_fif(directory / "sample_audvis_raw.fif", preload=False)
raw.pick_types(eeg=True)
raw.crop(0, 10)
raw.load_data()
raw.filter(1., 40.)
ica = ICA(n_components=None)
ica.fit(raw)
ica.plot_components(inst=raw)
```
```

### Expected results

-

### Actual results

-

### Additional information

-

drammock, 2023-05-12:
the plots are not created within a `plt.ion()` context so they're blocking execution.  A solution would be to generate (but not show) the figures in the for-loop over subsets of components, then show all of them at the end after the for-loop completes.  Getting this stuff to work right in all contexts (notebook, ipython, spyder, pycharm) is tricky; don't hesitate to ask for help if you get stuck.

mscheltienne, 2023-05-15:
More discoveries! In #11696 I removed the recursion in `plot_ica_components` and now gather all figures and show only once at the end. I don't see a downside in this approach, and it solves part of the issue. All the figures open at once. 

But:
- In a python interpreter, works great.
- In an IPython console, works great.
- In a Spyder console or in a jupyter notebok, closing the figure hangs the interpreter :face_exhaling: 

Speedy screencast to demonstrate (too bad it doesn't capture the mouse..)

[Screencast from 05-15-2023 02:46:05 PM.webm](https://github.com/mne-tools/mne-python/assets/73893616/85ddc383-6745-4208-b752-6006d2702efd)

And now the discovery part. It works on user-provided axes!

```
from matplotlib import pyplot as plt
from mne.datasets import sample
from mne.io import read_raw_fif
from mne.preprocessing import ICA

directory = sample.data_path() / "MEG" / "sample"
raw = read_raw_fif(directory / "sample_audvis_raw.fif", preload=False)
raw.pick_types(eeg=True)
raw.crop(0, 10)
raw.load_data()
raw.filter(1., 40.)
ica = ICA(n_components=5)
ica.fit(raw)

f, ax = plt.subplots(1, 2)
ica.plot_components(inst=raw, picks=[0, 2], axes=ax)
```

This works in all of those 4 tests and does not hang the interpreter in Spyder and Jupyter. Something is not behaving around here.. https://github.com/mne-tools/mne-python/blob/1e6aa7d99530f4cd11505665307b5667fe126b8a/mne/viz/utils.py#L470-L527

mscheltienne, 2023-05-15:
https://github.com/mne-tools/mne-python/blob/1e6aa7d99530f4cd11505665307b5667fe126b8a/mne/viz/_mpl_figure.py#L78-L85

What is the purpose of those lines? If we remove them, everything works fine in python, ipython, spyder and jupyter (still needs to be tested in VSCode, PyCharm).

PR #11696 (2023-05-16): Remove recursion in plot_ica_components and use context manager for plt.ion/plt.ioff
Closes #11693
Closes #11510 (lucky hit, that was the same bug at play)

Prevents spyder console and jupyter notebook from hanging because of:

```
plt.ion()
# stuff
plt.ioff()
```

And restructure `plot_ica_components` to remove recursion and gather all figures before `plt.show`.
closed 2023-05-16 by larsoner