mne-tools/mne-python #11699: Raw object plotting bug?

kulasbart, 2023-05-23:
### Description of the problem

When using the raw object plot method to review the data, I get a blocky barcode looking plot:

[image]

All I have done here is:
- loaded and concatenated some eeg data in csv form
- downsampled to standard 19 chan 10-20
- created raw mne object, set montage to standard_1020, and plotted

I have included the code to reproduce this error. I'm curious because the plot seems to work fine if I don't pass anything for ch_types in mne.create_info().

### Steps to reproduce

```Python
import pandas as pd
import glob
import os
import numpy as np
import mne 
from matplotlib import pyplot as plt

dfs = []
for i_csv in sorted(glob.glob('*.csv')):
    print(i_csv)
    dfs.append(pd.read_csv(i_csv))
    
df = pd.concat(dfs)
df.loc[df['sensor position']=='FP1', 'sensor position'] = 'Fp1'
df.loc[df['sensor position']=='FP2', 'sensor position'] = 'Fp2'
df.loc[df['sensor position']=='CZ', 'sensor position'] = 'Cz'
df.loc[df['sensor position']=='PZ', 'sensor position'] = 'Pz'
df.loc[df['sensor position']=='FZ', 'sensor position'] = 'Fz'

eeg_data = {}

for sensor in df['sensor position'].unique():
    eeg_data[sensor] = df[df['sensor position']==sensor]['sensor value'].values

chans_1020 = ['Fp1', 'Fp2', 'F7', 'F3', 'Fz', 'F4', 'F8', 'C3', 'Cz', 
              'C4', 'P3', 'Pz', 'P4', 'T7', 'T8', 'P7', 'P8', 'O1', 'O2']

eeg = []
channel_names = []
for chan in eeg_data.keys():
    if chan in chans_1020:
        channel_names.append(chan)
        eeg.append(eeg_data[chan])

sf = 256
mne_info = mne.create_info(channel_names,
                           sf,
                           ch_types=['eeg']*len(channel_names)
                          )

mne_raw = mne.io.RawArray(data, mne_info)

standard_1020 = mne.channels.make_standard_montage('standard_1020')
mne_raw.set_montage(standard_1020)

# interactive plot
%matplotlib ipympl
mne_raw.plot();
```

### Link to data

[Data1.csv](https://github.com/mne-tools/mne-python/files/11539944/Data1.csv)
[Data2.csv](https://github.com/mne-tools/mne-python/files/11539945/Data2.csv)
[Data3.csv](https://github.com/mne-tools/mne-python/files/11539946/Data3.csv)
[Data4.csv](https://github.com/mne-tools/mne-python/files/11539947/Data4.csv)
[Data5.csv](https://github.com/mne-tools/mne-python/files/11539948/Data5.csv)
[Data6.csv](https://github.com/mne-tools/mne-python/files/11539949/Data6.csv)
[Data7.csv](https://github.com/mne-tools/mne-python/files/11539950/Data7.csv)
[Data8.csv](https://github.com/mne-tools/mne-python/files/11539951/Data8.csv)

### Expected results

(without passing 'eeg' for ch_types)

[image]

### Actual results

[image]

### Additional information

Platform:         macOS-10.16-x86_64-i386-64bit
Python:           3.9.7 (default, Sep 16 2021, 08:50:36)  [Clang 10.0.0 ]
Executable:       /Users/bartek/opt/anaconda3/bin/python
CPU:              i386: 8 cores
Memory:           8.0 GB

mne:              1.1.1
numpy:            1.21.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:            1.7.3
matplotlib:       3.5.2 {backend=module://ipympl.backend_nbagg}

sklearn:          1.1.1
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.2.0 {PyQt5=5.9.7}
ipympl:           0.9.1
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     0.4

kulasbart, 2023-05-23:
Some additional info on the eeg data used:
- https://archive.ics.uci.edu/ml/datasets/eeg+database
- https://www.kaggle.com/datasets/nnair25/Alcoholics

agramfort, 2023-05-23:
Your data is not scaled correctly. Make sure it’s on Volts and otherwise use scaling auto mode in raw.plot

mscheltienne, 2023-05-23:
Closing as this is not a bug, but a scaling issue. MNE expects data in SI units. You can find here the list of data types with the scaling applied by raw.plot: https://mne.tools/stable/glossary.html#term-data-channels Thus, for EEG, you should provide data in Volts and raw.plot will automatically scale by 1e+06 to plot in uV. Also, in a data browser created with raw.plot, you can rescale interactively with the +/- keys on your keyboard.

closed 2023-05-23 by mscheltienne