mne-tools/mne-python #11706: mne.io.read_raw_fil should support datasets without sensor positions

RCTimms, 2023-05-24:
### Describe the new feature or enhancement

It would be great to generalise mne.io.read_raw_fil so that it would read in data which didn't have channel positions, or the documentation should reflect that the positions.tsv file is required to use this method.

### Describe your proposed implementation

The existing method can be used, but should have a case to handle an instance of this file missing.

### Describe possible alternatives

n/a

RCTimms, 2023-05-24:
@georgeoneill

larsoner, 2023-05-24:
Maybe a new `on_pos_missing='raise'` would be good. Then set the loc to nan if no positions provided

georgeoneill, 2023-05-24:
This should be pretty easy to implement. The code already assumes there is [no position information for any sensors](https://github.com/mne-tools/mne-python/blob/4ce984f4458652551e2997da3749c2cbdb92a64e/mne/io/fil/fil.py#LL102C39-L102C39), but for some reason I forgot to get to check if position files exists in the current iteration of the code.

RCTimms, 2023-05-24:
Thanks for looking into this

PR #11733 (2023-06-23): [MNT] Add support for FIL data without sensor positions
#### Reference issue
Fixes #11706

#### What does this implement/fix?
Allows FIL format data to be read in without supplying any sensor positions. 

#### Additional information
Currently it raises a warning if no sensor positions can be found but continues to import. Is there a smarter way to raise this?
closed 2023-06-23 by larsoner