mne-tools/mne-python #11710: Issue with read_raw_snirf function

wybiacx, 2023-05-28:
### Description of the problem

Hello MNE developers,

I've been using the MNE Python library for processing my fNIRS data in the SNIRF format. Recently, I encountered an issue while using the read_raw_snirf function to load my .snirf files. I receive an `IndexError: index 1 is out of bounds for axis 1 with size 1` during the loading process, even though I believe my files are correctly formatted.

I've tried using the snirpy library to load my data, and it seems to be able to read the files without issue. This leads me to believe there might be a potential problem with the read_raw_snirf function.

Here's a simplified version of the code I'm using:

```
from mne.io import read_raw_snirf
def load_data(fnirs_raw_dir):
    raw_intensity = read_raw_snirf(fnirs_raw_dir).load_data()
    return raw_intensity
```
I'm using MNE version `mne1.4.0` on Python 3.8 in a Windows11 environment.

Any guidance on how to resolve this issue would be much appreciated.

Thank you in advance for your help!

Best regards,
WYBIACX

### Steps to reproduce

```Python
from mne.io import read_raw_snirf
def load_data(fnirs_raw_dir):
    raw_intensity = read_raw_snirf(fnirs_raw_dir).load_data()
    return raw_intensity
data_path = './snirf/Block_2D_C.snirf'
    data = load_data(data_path)
```
```

### Link to data

[Block_2D_C.zip](https://github.com/mne-tools/mne-python/files/11583653/Block_2D_C.zip)
The link above is to the data I am using.

### Expected results

The read_raw_snirf function should be able to correctly load the data from my .snirf files without any errors.

### Actual results

(snirf) PS D:\Project\Snirf> python .\new_process.py
Processing file: Block_2D_C.snirf
Loading D:\Project\Snirf\snirf\Block_2D_C.snirf
Traceback (most recent call last):
  File ".\new_process.py", line 44, in <module>
    data = load_data(os.path.join(data_path, file))
  File ".\new_process.py", line 11, in load_data
    raw_intensity = read_raw_snirf(fnirs_raw_dir).load_data()
  File "D:\ProgramData\Miniconda3\envs\snirf\lib\site-packages\mne\io\snirf\_snirf.py", line 54, in read_raw_snirf
    return RawSNIRF(fname, optode_frame, preload, verbose)
  File "<decorator-gen-318>", line 12, in __init__
  File "D:\ProgramData\Miniconda3\envs\snirf\lib\site-packages\mne\io\snirf\_snirf.py", line 473, in __init__
    annot.append(data[:, 0], data[:, 1], desc.decode("UTF-8"))
IndexError: index 1 is out of bounds for axis 1 with size 1

### Additional information

Platform             Windows-10-10.0.22621-SP0
Python               3.8.16 (default, Mar  2 2023, 03:18:16) [MSC v.1916 64 bit (AMD64)]
Executable           D:\ProgramData\Miniconda3\envs\snirf\python.exe
CPU                  Intel64 Family 6 Model 154 Stepping 3, GenuineIntel (16 cores)
Memory               Unavailable (requires "psutil" package)
Core
├☑ mne               1.4.0
├☑ numpy             1.24.3 (MKL 2023.1-Product with 12 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=QtAgg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ pandas            1.5.3
└☐ unavailable       numba, dipy, openmeeg, cupy

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, ipyvtklink, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser

Ecosystem (optional)
├☑ mne-nirs          0.5.0
└☐ unavailable       mne-bids, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2023-05-28:
@rob-luke can you look?

rob-luke, 2023-05-28:
I'll take a look 👍

@wybiacx Where did this file come from? What machine did you acquire this data with (artinis, nirx, etc)? Have you processed the file with any other software?

HanBnrd, 2023-08-23:
Hi, we had the same problem after editing events in Homer. I believe this happens when a *snirf* file has some empty *stim* channels.

HanBnrd, 2023-08-23:
This could help fix empty *stim* channels in your *snirf* file:
**⚠️ this will overwrite your *snirf* file, make sure to back it up beforehand ⚠️**
```python
import h5py

file_name = 'my_file.snirf'
dat = h5py.File(file_name, 'a')
print(f"Original channels: {list(dat['nirs'].keys())}")
deleted = []
for k in dat['nirs'].keys():
    if 'stim' in k:
        if dat['nirs'][k]['data'].shape == (1, 1):
            deleted.append(dat['nirs'][k]['name'][:].tolist())
            del dat['nirs'][k]
print(f"Deleted: {deleted}")
print(f"New channels: {list(dat['nirs'].keys())}")
```

HanBnrd, 2023-08-23:
Just checked on the [*snirf* specifications](https://github.com/fNIRS/snirf/blob/master/snirf_specification.md#nirsistimjdata) and it says `stim.data` should be 2D and have at least 3 columns.

I suggest we change [this](https://github.com/mne-tools/mne-python/blob/18aa019be069445a2bdb413a57089ea670b8446f/mne/io/snirf/_snirf.py#L474) by this `if data.shape[1] >= 3 :` what do you think @rob-luke @larsoner ?

PR #11915 (2023-08-24): Fix snirf loader
#### Reference issue
Fixes #11710

#### What does this implement/fix?
Instead of checking that the `stim.data` size is strictly greater than 0, we check that it has at least 3 columns. This would follow the [*snirf* specifications](https://github.com/fNIRS/snirf/blob/master/snirf_specification.md#nirsistimjdata) about `stim.data`:
> This is a numeric 2-D array with at least 3 columns, specifying the stimulus time course for the jth condition. Each row corresponds to a specific stimulus trial. The first three columns indicate `[starttime duration value]`.

@rob-luke @larsoner
closed 2023-08-24 by larsoner