mne-tools/mne-python #11711: Baseline must be a tuple?

cbrnr, 2023-05-29:
I was wondering if forcing the `baseline` argument to be a tuple is necessary and/or added on purpose in #9210. I usually tend to use lists, and this used to work before this change. Now, a tuple is required. @hoechenberger? cf https://github.com/cbrnr/mnelab/issues/381

hoechenberger, 2023-05-29:
We always explicitly requested `baseline` to be a tuple as per the API docs, so I suppose this change you mentioned is just to ensure that the output (in the `.baseline` attribute) equals the input provided by the user. I think it'd be good to support `array-like of length 2` as input and cast to a tuple internally

cbrnr, 2023-05-29:
Yes, I'd also prefer list-like with two elements if there is no strong reason to require a tuple. If everyone agrees I can submit a PR.

agramfort, 2023-05-29:
Works for me

closed 2023-05-31 by hoechenberger