mne-tools/mne-python #11717: Plotting a psd turns off the interactive mode in matplotlib, which cause ica-plots to hang

gronm1, 2023-05-31:
### Description of the problem

raw.compute_psd().plot() turns of interactive mode, which cause some plots, like ica-plots to hang.

### Steps to reproduce

```Python
import mne
import matplotlib.pyplot as plt
import matplotlib

print("\n\nmatplotlib backend:",matplotlib.get_backend())
print("Is interactive:",plt.isinteractive(),"\n\n")

#%%
import os
from mne.preprocessing import ICA, corrmap, create_ecg_epochs, create_eog_epochs

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(
    sample_data_folder, "MEG", "sample", "sample_audvis_filt-0-40_raw.fif"
)
raw = mne.io.read_raw_fif(sample_data_raw_file)

# Here we'll crop to 60 seconds and drop gradiometer channels for speed
raw.crop(tmax=60.0).pick_types(meg="mag", eeg=True, stim=True, eog=True)
raw.load_data()

#%% preemptively turning on interactive mode does nothing
plt.ion()

#%%
raw.compute_psd().plot()

#%%
print("\n\nmatplotlib backend:",matplotlib.get_backend())
print("Is interactive:",plt.isinteractive(),"\n\n")

#%%
# !!!!!!!!!! CRUDE FIX: turn back on interactive mode in matplotlib !!!!!!!!!!
#plt.ion()

#%%
print("\n\nmatplotlib backend:",matplotlib.get_backend())
print("Is interactive:",plt.isinteractive(),"\n\n")

#%%
ica=mne.preprocessing.ICA(
    n_components=15,method='picard',max_iter=600,random_state=42
).fit(raw)
ica.plot_sources(raw)
ica.plot_components()

# !!!!!!!!!! python hangs here !!!!!!!!!!
print("\n\nThis part is not reached if interactive mode is off!")
```

### Expected results

Script running until end.

### Actual results

Scripts hangs on ica.plot_components() and closing the window does not help.

### Additional information

% mne sys_info                                  
Platform             Linux-5.15.0-72-generic-x86_64-with-glibc2.31
Python               3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:39:03)  [GCC 11.3.0]
Executable           /m/home/home0/03/gronm1/unix/micromamba/envs/visual_gamma_mne_1.4.0/bin/python3.9
CPU                  x86_64 (16 cores)
Memory               31.0 GB

Core
├☑ mne               1.4.0
├☑ numpy             1.24.3 (unknown linalg bindings)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=QtAgg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.57.0
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ cupy              12.0.0
└☑ pandas            2.0.2

Visualization (optional)
├☑ pyvista           0.39.1 (OpenGL 4.5.0 NVIDIA 470.182.03 via NVIDIA RTX A2000 12GB/PCIe/SSE2)
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.3
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.5.0
└☐ unavailable       ipympl

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

---

Extra observation:
Running this issue-reproducing script in IPython once does cause the issue. However, then interrupting the kernel and rerunning this script does not cause the issue again.

larsoner, 2023-05-31:
This should be fixed on the latest dev version, we should probably release 1.4.1 with the fix!

hoechenberger, 2023-09-24:
@larsoner I believe this can be closed

closed 2023-09-24 by larsoner