mne-tools/mne-python #11722: Compute and plot PSD for all channels

cbrnr, 2023-06-06:
Our definition of `picks` is inconsistent across various functions (see e.g. #11382). Here's another example where we could make the behavior more consistent: Plotting the PSD of *all* channels (good and bad).

Currently, this is not possible, or at least it is not immediately obvious how to do it. I've described this in our forum: https://mne.discourse.group/t/computing-and-plotting-psd-with-all-channels/7025/1

In a nutshell:

- The only way to include bad channels in `psd = raw.compute_psd()` is to explicitly list them in `picks`. Rather unintuitively, `picks="all"`, `picks="data"`, and `picks="eeg"` all include only *good* channels.
- In addition, plotting a `Spectrum` object with `psd.plot()` has its own `picks` parameter. But if bad channels are not present in the first place, the recent change in v1.5 of showing all channels by default has no impact.

Here's a possible solution to improve consistency:

- Change `picks="all"` to use *all* channels (good and bad)
- Add `picks="good"` to use only *good* channels
- Add `picks="bad"` to use only *bad* channels
- Change `picks="data"` to use *data* channels (good and bad)
- Change `picks="eeg"` (and other types) to use *all* channels of that type (good and bad)

However, this gets a bit complicated, especially when considering that at least `Spectrum.plot()` has an `exclude` parameter (which is basically the opposite of `picks`). I assume that only one of the two arguments can be passed, but we'd also need to translate any changes to the behavior of `picks` to `exclude`. BTW, `raw.compute_psd()` has no `exclude` parameter, which is also inconsistent.

Another option could be to add a new parameter to deal with bad channels (e.g. `include_bads=True`). This would make the two selections independent of each other (and probably simpler, IDK).

larsoner, 2023-06-06:
From what I recall in several places we have `picks=...` *paired with* `exclude=...` so you could do for example `picks='all', exclude='bads'` to pick all non-bad channels, `picks='data', exclude=()` to pick all good data channels, etc. So maybe all we're missing from some functions (these included) would be an `exclude` param which says what to exclude as well.

cbrnr, 2023-06-06:
If that's how it's already done in some functions I'm OK with having a combination of `picks` and `exclude`. By adding `exclude` to `raw.compute_psd()`, we can make the current default behavior explicit (`picks="all", exclude="bads"`) or, even better, make the new default consistent with other places (`picks="all", exclude=()`).

cbrnr, 2023-06-06:
But then how do we handle things like `picks=["C3", "Cz", "C4"], exclude=["C3", "Cz", "C4"]`? Do we just take the difference `picks - exclude` of those sets?

larsoner, 2023-06-06:
> or, even better, make the new default consistent with other places (picks="all", exclude=()).

I *think* @drammock was actually trying to move this direction recently with some `Spectrum` class changes so let's wait for him to comment

> But then how do we handle things like picks=["C3", "Cz", "C4"], exclude=["C3", "Cz", "C4"]? Do we just take the difference picks - exclude of those sets?

No idea how this is currently handled, `exclude` might only be used for str-type `picks` rather than lists. Feel free to check and document somewhere

agramfort, 2023-06-08:
I would copy how evoked.plot behaves
https://mne.tools/stable/generated/mne.Evoked.html#mne.Evoked.plot

> Message ID: ***@***.***>
>

larsoner, 2023-09-12:
Sounds good to me

closed 2023-10-04 by drammock