mne-tools/mne-python #11726: mne.viz.plot_topomap with bipolar montage

NoPenguinsLand, 2023-06-09:
### Description of the problem

Could be considered bug or enhancement.

I'm re-referencing my EEG data to use bipolar montage (using mne.io.set_bipolar_reference) and am using the same channels multiple times as anodes with different cathodes (e.g. Fp1-AF3 and Fp1-AF7). The way mne currently works is to set the new virtual channel's information including sensor location the same as the anode's. As a result, both Fp1-AF3 and Fp1-AF7 virtual channels will have the same sensor locations. No error is actually being raised at this point.

The error will be raised when I tried calling mne.viz.plot_topomap and it'll yield the following message:

`ValueError: The following electrodes have overlapping positions, which causes problems during visualization:`

Two changes as a possible solution:

1. Add update_loc boolean argument to set_bipolar_reference function to set the new virtual channel's sensor location as the midpoint of anode and cathode's sensor locations.
2. Raise warning if multiple channels of the same type have the same sensor locations during set_bipolar_reference.

### Steps to reproduce

```Python
import matplotlib as mpl

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = (
    sample_data_folder / "MEG" / "sample" / "sample_audvis_filt-0-40_raw.fif"
)
raw = mne.io.read_raw_fif(sample_data_raw_file, preload=True)
info = mne.io.read_info(sample_data_raw_file)
raw.pick(mne.pick_types(info, meg=False, eeg=True, exclude=[]))
anode = ['EEG 004','EEG 005','EEG 004']
cathode = ['EEG 006','EEG 007','EEG 008']

mne.set_bipolar_reference(raw, anode=anode, cathode=cathode, copy=False, drop_refs=True)
raw.drop_channels(['EEG 001', 'EEG 002', 'EEG 003', 'EEG 009', 'EEG 010', 'EEG 011', 'EEG 012', 'EEG 013', 'EEG 014', 'EEG 015', 'EEG 016', 'EEG 017', 'EEG 018', 'EEG 019', 'EEG 020', 'EEG 021', 'EEG 022', 'EEG 023', 'EEG 024', 'EEG 025', 'EEG 026', 'EEG 027', 'EEG 028', 'EEG 029', 'EEG 030', 'EEG 031', 'EEG 032', 'EEG 033', 'EEG 034', 'EEG 035', 'EEG 036', 'EEG 037', 'EEG 038', 'EEG 039', 'EEG 040', 'EEG 041', 'EEG 042', 'EEG 043', 'EEG 044', 'EEG 045', 'EEG 046', 'EEG 047', 'EEG 048', 'EEG 049', 'EEG 050', 'EEG 051', 'EEG 052', 'EEG 053', 'EEG 054', 'EEG 055', 'EEG 056', 'EEG 057', 'EEG 058', 'EEG 059', 'EEG 060'])

mpl.use('QtAgg')

fig, axs = plt.subplots(figsize=(7.5, 4.5), nrows=2, ncols=2)

mne.viz.plot_topomap(raw, pos=raw.info, axes=axs[0,0], show=True)
```
```

### Expected results

None

### Actual results

```
ValueError: The following electrodes have overlapping positions, which causes problems during visualization:
EEG 004-EEG 006, EEG 004-EEG 008
```

### Additional information

Platform:         Windows-10-10.0.19045-SP0
Python:           3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:00:38) [MSC v.1934 64 bit (AMD64)]
Executable:       C:\Users\mning\AppData\Local\mambaforge\envs\mne\python.exe
CPU:              Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:           7.8 GB

mne:              1.3.1
numpy:            1.23.5 {MKL 2022.1-Product with 4 threads}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=QtAgg}

sklearn:          1.2.2
numba:            0.56.4
nibabel:          5.1.0
nilearn:          0.10.1
dipy:             1.7.0
openmeeg:         2.5.6
cupy:             Not found
pandas:           2.0.1
pyvista:          0.39.0 {OpenGL 4.5.0 - Build 30.0.101.1122 via Intel(R) UHD Graphics 620}
pyvistaqt:        0.0.0
ipyvtklink:       0.2.2
vtk:              9.2.6
qtpy:             2.3.1 {PyQt5=5.15.6}
ipympl:           Not found
pyqtgraph:        0.13.3
pooch:            v1.7.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.0.0
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2023-06-10:
honestly I don't know what is the sound behavior to adopt here. Sorry.

> Message ID: ***@***.***>
>

NoPenguinsLand, 2023-06-20:
Good idea. @agramfort should I submit PR just for the warning?

FWIW, mne-nirs handles this by merging overlapping channels/excluding channels with redundant locations during call to plotting functions as shown [here](https://github.com/mne-tools/mne-nirs/blob/5f005d710ddf529f7cfd1d809b2d9f4383e07a48/mne_nirs/visualisation/_plot_GLM_topo.py#L292) which leads to [here](https://github.com/mne-tools/mne-python/blob/b0f8ce69c7258319624aebdd6de63b23e1c78ad5/mne/viz/topomap.py#L180), which is less than ideal.

drammock, 2023-06-20:
maybe a naive question, but can you simply switch cathodes and anodes, to therefore end up with distinct locations, and therefore be able to plot?  I'm not saying it's electromagnetically the most accurate approach, but depending on what you want to learn from the plot it might be good enough?

NoPenguinsLand, 2023-06-20:
Same error will be raised for O1 and O2 channels.
[image]

NoPenguinsLand, 2023-06-21:
> this is not a solution per se, but given the cathode/anode pairs that I
> see in that plot, I wonder if something like current source density
> <https://mne.tools/stable/auto_examples/preprocessing/eeg_csd.html> would
> be an alternative that interests you @NoPenguinsLand
> <https://github.com/NoPenguinsLand> ?

It's not really in my current plan tbh 😂 but thank you.