mne-tools/mne-python #11728: Issue in brain.add_volume_labels()

Dr-Chen-Xiaoyu, 2023-06-11:
### Description of the problem

Hi,

I am trying to go through the code of [Working with sEEG data — MNE 1.4.2 documentation](https://mne.tools/stable/auto_tutorials/clinical/20_seeg.html). However, when I am using “brain.add_volume_labels”, I found that the plot is different bewteen my Linux server and my laptop (windows system).

### Steps to reproduce

```Python
%matplotlib qt
import mne
mne.viz.set_3d_backend('pyvistaqt') # GUI requires pyvista backend
mne.viz.set_3d_options(depth_peeling=False, antialias=False, multi_samples=1) # for bugs of pure black plot of 3D viz

fsaverage_dir="/data/xyc/codes/Tests/MNE_fetch_fsaverage"
mne.datasets.fetch_fsaverage(subjects_dir=fsaverage_dir, verbose=True)  # use mne-python's fsaverage data
mne.datasets.fetch_hcp_mmp_parcellation(subjects_dir=fsaverage_dir,verbose=True) # use mne-python's hcp mmp parcellation

brain = mne.viz.Brain(
    "fsaverage",
    subjects_dir=fsaverage_dir,
    cortex='low_contrast', alpha=0.2, background='white'
)

brain.add_volume_labels(aseg="aparc+aseg")
```

### Expected results

The brain from fsaverage and its parcellation should be correctly aligned for both linux and windows.

### Actual results

For windows, the same code (only change the path of fsaverage_dir) is Okay with correct plot. 
However, Linux’s plot is wrong. Note that I run my code with jupyter notebook in a remote Linux server. %matplotlib qt can run well with mne’s 3D plot. This plot also runs well with no errors or warmings.

Plots can be found on https://mne.discourse.group/t/issue-in-brain-add-volume-labels/7037. Since [Alexandre Gramfort](https://mne.discourse.group/u/agramfort) also replicated this isssue, I think maybe it is some bug or mistake in brain.add_volume_labels() that the coordinates are misused in Linux. 

Best，
Xiaoyu😊

### Additional information

My outputs of “mne sys_info” for windows and Linux:

```
Platform             Linux-5.19.0-41-generic-x86_64-with-glibc2.35
Python               3.11.3 | packaged by conda-forge | (main, Apr  6 2023, 08:57:19) [GCC 11.3.0]
Executable           /home/xyc/anaconda/envs/env_mne/bin/python3.11
CPU                  x86_64 (128 cores)
Memory               1007.5 GB

Core
├☑ mne               1.4.0
├☑ numpy             1.24.3 (unknown linalg bindings)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.57.0
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.1
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.39.1 (OpenGL 4.5 (Core Profile) Mesa 22.2.5 via llvmpipe (LLVM 15.0.6, 256 bits))
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.5.0
└☐ unavailable       ipympl

Ecosystem (optional)
├☑ mne-bids          0.12
└☐ unavailable       mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline
```

```
Platform:         Windows-10-10.0.22621-SP0
Python:           3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:28:38) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\XYchen\anaconda3\envs\envmne\python.exe
CPU:              AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD: 16 cores
Memory:           13.9 GB

mne:              1.3.1
numpy:            1.23.5 {MKL 2022.1-Product with 8 threads}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.2.1
numba:            0.56.4
nibabel:          5.0.1
nilearn:          0.10.0
dipy:             1.6.0
openmeeg:         2.5.5
cupy:             Not found
pandas:           1.5.3
pyvista:          0.38.3 {OpenGL 4.6.0 Compatibility Profile Context 22.20.44.221025 via AMD Radeon(TM) Graphics}
pyvistaqt:        0.9.1
ipyvtklink:       0.2.2
vtk:              9.2.6
qtpy:             2.3.0 {PyQt5=5.15.6}
ipympl:           Not found
pyqtgraph:        0.13.2
pooch:            v1.7.0

mne_bids:         0.12
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found
```

larsoner, 2023-06-12:
@alexrockhill can you replicate https://mne.discourse.group/t/issue-in-brain-add-volume-labels/7037 locally?

alexrockhill, 2023-06-12:
I'll look into this in 3-4 hours.

closed 2023-06-14 by larsoner