mne-tools/mne-python #11731: I/O roundtrip for annotations from a cropped raw induces a shift of raw.first_samp samples

mscheltienne, 2023-06-13:
### Description of the problem

If you crop a `raw` (with no measurement date), inducing a change in `raw.first_samp`, then save and reload the annotations attached, they are shifted by `raw.first_samp`. Let's consider a `raw` recording which has never been cropped and starts at `raw.first_samp = 0`, with 2 annotations at 2 and 4 seconds lasting 0.5 seconds:

```
from mne import Annotations, read_annotations
from mne.datasets import sample
from mne.io import read_raw_fif

fname = sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = read_raw_fif(fname, preload=False)
raw.pick_types(eeg=True)
raw.crop(0, 10, include_tmax=True)
raw.load_data()
raw.set_meas_date(None)  # remove measurement date since it impacts annotations
raw.set_annotations(None)
raw._cropped_samp = 0  # overwrite first_samp to 0 for this example
annots = Annotations(onset=[2, 4], duration=0.5, description="test", orig_time=None)
raw.set_annotations(annots)
raw.plot(n_channels=5)
```

[image]

Now let's crop the first second and set `raw.first_samp` to the `601` sample. 

```
raw.crop(1, 8, include_tmax=True)
raw.plot(n_channels=5)
```

[image]

And finally, let's save the annotations to a separate file and load them back:

```
fname = "~/Downloads/test_annot.fif"
raw.annotations.save(fname, overwrite=True)
annots = read_annotations(fname)
raw.set_annotations(annots)  # overwrite existing annotations
raw.plot(n_channels=5)
```

[image]

-----------------------

By the end of the I/O roundtrip, the annotations are shifted by `raw.first_samp`, in this case 1 second. The "fix" is to edit the `.onset` attribute:

```
fname = "~/Downloads/test_annot.fif"
raw.annotations.save(fname, overwrite=True)
annots = read_annotations(fname)
annots.onset -= raw.first_samp / raw.info["sfreq"]
raw.set_annotations(annots)  # overwrite existing annotations
raw.plot(n_channels=5)
```

But `Annotation.onset` is not documented, I don't expect users to edit this attribute.

-----------------------

In the `Notes` of the `Annotations` class, there is this warning:

> This means that when raw.info['meas_date'] is None, doing raw.set_annotations(raw.annotations) will not alter raw if and only if raw.first_samp == 0. When it’s non-zero, raw.set_annotations will assume that the “new” annotations refer to the original data (with first_samp==0), and will be re-referenced to the new time offset!

which seems related and is completely unclear to me. 

> raw.set_annotations(raw.annotations) will not alter raw [...]

Alter raw? How would that alter it if the `raw.first_samp` is different from 0?

-----------------------

Anyway, IMO, this I/O roundtrip should just work without users requiring to edit undocumented attributes of the re-loaded `Annotations`.

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

mscheltienne, 2023-06-13:
pinging @ArthurNguyen if you want to follow.

larsoner, 2023-06-13:
> Anyway, IMO, this I/O roundtrip should just work without users requiring to edit undocumented attributes of the re-loaded Annotations.

Agreed in principle but in practice IIRC this would break other behaviors where we have thought "it should work X way in Y case", so it's not so easy. I suspect changing the behavior as you suggest would be backward incompatible and break those other use cases. I don't know what they are offhand and always have to read the code/tests extensively to remember :disappointed: We do have a lot of tests for this stuff so you could modify the code to do what you suggest but you'll see it breaks stuff...

I think we should:

1. Document this stuff properly / better and edit the parts that don't make sense.
2. Consider adding some new kwarg like `raw.set_annotations(..., whatever='current behavior')` that defaults to our current behavior but you could set to `whatever='something else'` to get the behavior you'd expect that could work (for suitable `whatever`, `current behavior`, and `something else`). Some naming having to do with `meas_date` being None *and* the instance being cropped *and* saving and loading annotations, since I think that's the only time this comes into play...

agramfort, 2023-06-17:
I don't see an easy way to support this usecase but I would look into a way
to rely on orig_time that is not None to make sure everything is always
aligned wrt some absolute clock time

> Message ID: ***@***.***>
>

mscheltienne, 2023-06-19:
> I suspect changing the behavior as you suggest would be backward incompatible and break those other use cases. 

Very likely.. Anyway, the documentation should be improved. I will also try a couple of changes when I can, see if I can fix this without breaking anything.