mne-tools/mne-python #11743: Subspace pursuit sparse source localisation

jshanna100, 2023-06-19:
### Describe the new feature or enhancement

Hi, a while ago I implemented the the subspace pursuit algorithm for sparse source localisation described here:
https://www.sciencedirect.com/science/article/abs/pii/S1053811913009440

I wonder if there is interest in adding it to the already existing suite of tools for sparse source localisation?

### Describe your proposed implementation

The branch is found here, if memory serves me correctly I followed the paper pretty closely:
https://github.com/jshanna100/mne-python/tree/greedy

### Describe possible alternatives

The authors of the paper do state that their code is available on request, but I never had any luck getting it from them, which is why I did my own implementation.

larsoner, 2023-06-20:
@agramfort WDYT about adding subspace pursuit?

agramfort, 2023-06-23:
no strong feeling here. I just fear that the interaction with a hierarchical source space is a bit harder than the other solvers we have.