mne-tools/mne-python #11747: Wavelet-ICA implementation?

Hororohoruru, 2023-06-22:
### Describe the new feature or enhancement

Wavelet-ICA (wICA) as described in:

N.P. Castellanos, and V.A. Makarov (2006). 'Recovering EEG brain signals: Artifact suppression with wavelet enhanced independent component analysis' J. Neurosci. Methods, 158, 300--312.

### Describe your proposed implementation

I imagine it could be included in mne.preprocessing, similar to `ICA`. I don't know how different it is from regular ICA for it to make sense to be a separate method or rather to be selected from `ICA` via parameters.

### Describe possible alternatives

Instead of a separate class, it could be implemented as a new `wavelet` boolean parameter of the `ICA` class. I believe a separate method is clearer, but maybe the process of wICA with respect to regular ICA makes it easy to implement in the aforementioned way.

agramfort, 2023-06-22:
do you know a python implementation for this?

Hororohoruru, 2023-06-27:
This is the only python implementation I found: https://github.com/compmem/ptsa/blob/master/ptsa/wica.py

agramfort, 2023-06-27:
It’s gpl licensed so you cannot copy the code like this. Mne only accepts bsd or mit licenses

Hororohoruru, 2023-06-27:
I understand. Is there any possibility of requesting it to be implemented in MNE in any other way?

agramfort, 2023-06-27:
you can ask the authors if they accept to relicense in BSD

> Message ID: ***@***.***>
>

hoechenberger, 2023-06-27:
If we keep the logic in

hoechenberger, 2023-06-27:
If the implementation remains in that external package and MNE implements an interface, the licensing question goes away. You could Interface with it the same way as we currently use e.g. Picard.

juangpc, 2023-06-27:
Hi, maybe this can help. http://www.mat.ucm.es/~vmakarov/research.php

Not sure about licenses, but citing properly it will probably be ok, but
perhaps you guys can ask him.

drammock, 2023-06-28:
> Not sure about licenses, but citing properly it will probably be ok

incorporating GPL-licensed code into a non-GPL-compatible codebase is not "probably OK". It is *definitely a violation of the GPL license*.  If you want wICA added to MNE-Python, the options are:

1. find an implementation (in Python or some other language) that has a compatible license (MIT, BSD...), and adapt/translate the code
2. find an implementation with a non-compatible license and get permission from the author(s) to re-license the code as BSD, then adapt/translate
3. re-implement the algorithm from first principles (i.e., looking at the equations in the published paper, but not looking at their code)

juangpc, 2023-06-28:
Not really following. The code is in matlab. It will need to get
redeveloped at least.
I already contacted the author.
I do not want anything. Just trying to help, both the author, and
mnepython.

drammock, 2023-06-28:
> Not really following. 

@juangpc what part was unclear?

> The code is in matlab.

It does not matter what language the code is written in. If it is GPL licensed, we need permission from its author to re-license it in order to legally incorporate it into MNE-Python (which does not have a GPL-compatible license; MNE's license is more permissive than GPL). see https://www.gnu.org/licenses/gpl-faq.en.html#TranslateCode

> It will need to get redeveloped at least.

that is why I said we would need to "adapt / translate the code".

juangpc, 2023-06-29:
Hi,
You're right, GPL code cannot be incorporated into a non-compatible license
codebase. The link provided is from the authors' publication and contains
code that is not GPL licensed. I was only trying to help find an example
implementation since I know the authors. Sorry if it caused any confusion.
Maybe the original creator of this issue can contact the authors and verify
any potential issues.

drammock, 2023-06-29:
ah, OK, thanks for clarifying!  I took a look at the `.zip` files containing the MATLAB code, and it appears that no license at all was specified, only a statement of "please acknowledge our work".  In this case (because of the [Berne Convention](https://en.wikipedia.org/wiki/Berne_Convention)) we still must contact the author and get permission to translate their code and release it with a BSD-3 license --- when no license is specified, no re-use rights at all are transferred to us.

Hororohoruru, 2023-06-29:
Hello all,

Thanks for the discussion and the suggestions. I will contact the original
author about the Matlab implementation that Juan shared.

Hororohoruru, 2023-07-01:
Hello again,

The author just replied to me and he granted us permssion to adapt the code from his Matlab example to python and publish it on mne under a BSD-3 license.

agramfort, 2023-07-02:
👌

sappelhoff, 2023-10-24:
> The author just replied to me and he granted us permssion to adapt the code from his Matlab example to python and publish it on mne under a BSD-3 license.

cool!

just for transparency it would be great to have this permission in this thread (preferably as a post directly be the author ... but maybe also a quoted email would be fine, if shared with permission)

christoph1212, 2026-04-09:
Hello all,
I found this thread on my search for a wICA MNE implementation. Is it now possible to use wICA in MNE?
Thanks and best wishes