mne-tools/mne-python #11749: Scrolling on our homepage

cbrnr, 2023-06-23:
Our homepage https://mne.tools/stable/index.html currently has two scrolling-related issues that I think should be fixed if possible:

1. The main contents takes up a lot of vertical space. It would be nice if this could be reduced a bit, because then the website would not require scrollbars on more devices. One idea is to try to get the Donders logo, which now occupies a third row on its own, in one of the two logo rows.
2. The left sidebar has scrollbars, at least on my Mac. In Safari, there are both horizontal and vertical scrollbars, whereas in Chrome there's only a vertical one. It looks like an issue with padding, because all of the content is already shown and there should be no need for scrollbars at all.

@drammock do you think this is fixable?

drammock, 2023-06-23:
re point 2: on linux/firefox there are no scrollbars on the left sidebar unless I shrink the browser height to where the sidebar quick links tile starts to be hidden. Please check if there's already an issue about safari scrollbars on https://github.com/pydata/pydata-sphinx-theme, and if not, raise one there

re point 1: both numpy and matplotlib have (much) longer homepages, scipy and pandas have (slightly) shorter ones. So I don't see us as being a major outlier here, and hesitate to put *too* much time into this. Easy-ish options I see are:

- shrink the tiles a bit.
- shrink the tiles a lot, and make it so that they are much smaller, and on hover the actual tile size zooms (not just the background image).  (this is not really easy-ish, but including it in case you're really motivated)
- combine the logo lists into one "supporting institutions" list (funders are technically "supporting institutions" too).

I hesitate to try your suggestion of squeezing the logos to get donders on the second line, seems like that just kicks the can down the road (in fact, #11741 added a new logo but forgot to actually display it on the homepage... so I think the "institutions fit on 2 lines" ship has sailed already)

drammock, 2023-06-23:
EDIT: #11741 is fine, I was looking at `stable` not `dev` docs.  `dev` already has *2* more logos than stable, so yeah, that ship has sailed.

cbrnr, 2023-06-28:
Re 1: I see, the dev version does look better, because a single logo on its own line seems a bit out of place. However, if the number of logos keeps increasing, we should have a plan for sometime in the future. Do we include logos of all past contributors? Or just for the currently active ones? How is this decided?

Re 2: The vertical scrollbar also appears with Chrome, so this is not a Safari problem. It seems to be [this issue](https://github.com/pydata/pydata-sphinx-theme/issues/1238), so I'll add a comment there.

drammock, 2023-06-28:
> if the number of logos keeps increasing, we should have a plan for sometime in the future.

I offered some ideas already in [my previous comment](https://github.com/mne-tools/mne-python/issues/11749#issuecomment-1605038414).

> Do we include logos of all past contributors? Or just for the currently active ones? How is this decided?

Logos reflect institutions of all present and past steering council members (or what used to be called "core devs").  In other words, anyone contributing enough to get commit rights gets their institution logo added.  On [this page](https://mne.tools/dev/funding.html) we split it out into "current" and "former" partner institutions, but on the homepage there is no distinction.

As far as how *was* this decided: I recall a discussion with @larsoner at some point in the past about whether the homepage logos should reflect only current steering council members but I can't turn up evidence of it (maybe it was in a dev meeting, not a PR?).  I'm willing to reconsider that decision if you want; please open a new issue about it if so.

cbrnr, 2023-06-28:
Thanks @drammock for the explanation. I saw your suggestions, but I think these are also only going to work up until a certain number of institutions (logos) is reached. At some point, having 50+ logos will probably be a bit overwhelming and we should think about a better way to present supporting institutions. But I guess this is still way too much in the future, so no need to start discussing this now!

Also, I did not want to discuss the mechanics of who gets to be on the homepage, I just didn't know how it works!