mne-tools/mne-python #11759: Tutorial for EEG-Eyetracking integration

scott-huberty, 2023-06-28:
### Proposed documentation enhancement

I would like to update the current eye-tracking tutorial to demonstrate integrating an eyetracking `RawEyelink` object with EEG  (in this case, a `RawMFF` object).

We don't need to add any new functions to MNE for this. We can use `mne.preprocessing.realign_raw`, passing in the photo-diode DIN events that were sent to both the Eyetracking and EEG acquisitions, to align the EEG and eye-tracking signals. I've been doing this for a study we're currently running and it works well!

I have an updated recording from the Pupillary Light Reflex (including both the eyelink and EGI files) that we can use for this.

larsoner, 2023-06-28:
Awesome, this would be a good thing to work on next! We need to add a new dataset for this, too, right?

scott-huberty, 2023-06-28:
> Awesome, this would be a good thing to work on next! We need to add a new dataset for this, too, right?

Yeah! I think our  tutorial's current PLR dataset is downloaded from OSF - so I can update it there (IIRC, My EEG file for the updated dataset is like 200mb.. So probably better to keep it on OSF and not in `mne-sample-datasets`?)

larsoner, 2023-06-29:
OSF is fine to use for datasets, feel free to just update the dataset there and we can add it as a new `mne.datasets.<something>`

PR #11770 (2023-07-06): DOC: EEG eye-tracking alignment tutorial
Closes #11759

- I updated the current eye-tracking tutorial to include a section on aligning and combining simultaneously collected EEG and eye-tracking data

- This required updating the eyelink sample dataset, including a companion EEG file ( ~ 100mb).

- UPDATE: here is the [tutorial](https://output.circle-artifacts.com/output/job/6d048d78-ba13-425f-ac13-bf4c216ba5fa/artifacts/0/html/auto_tutorials/preprocessing/90_eyetracking_data.html#sphx-glr-auto-tutorials-preprocessing-90-eyetracking-data-py)

EDIT: Hmmm... I can see that all my commits from PR [... cut]
closed 2023-07-06 by larsoner