mne-tools/mne-python #11767: psd welch calculated incorrectly at frequency=0

chapochn, 2023-06-30:
### Description of the problem

the mne function `psd_array_welch` uses the `scipy.signal.spectrogram` function for each segment. The latter function has the option `detrend` which is set as default to 'constant'. This signifies that the mean is subtracted because calculating the psd, which is most probably not desired and not documented in mne. I do not believe it would make sense as a default behavior. I suggest that the `detrend` option be set to "False" when calling spectrogram from `mne`, or at least given a choice.

### Steps to reproduce

```Python
import mne
import numpy as np
import matplotlib.pyplot as plt
import scipy.fft as sfft

n = 256
x = np.random.rand(n)-0.4 # not centered
fs = 100
psd1, f = mne.time_frequency.psd_array_welch(x, fs, window='boxcar')
dft = sfft.rfft(x)
psd2 = np.abs(dft * np.conjugate(dft)) / n /fs
psd2[1:] *= 2

plt.figure()
plt.plot(f, psd1, label='mne')
plt.plot(f, psd2, label='mine')
plt.legend()
plt.show()

print(psd1[:5])
print(psd2[:5])

output:
[1.50992908e-33 2.62355362e-03 1.04821458e-04 1.20104558e-03
 7.80199506e-04]
[0.03302601 0.00262355 0.00010482 0.00120105 0.0007802 ]
```

### Expected results

psd1[0] should be the same as psd2[0]

[image]

### Actual results

psd1[0] and psd2[0] are different

### Additional information

Platform             macOS-13.3.1-arm64-arm-64bit
Python               3.10.11 | packaged by conda-forge | (main, May 10 2023, 19:01:19) [Clang 14.0.6 ]
Executable           /Users/chapon01/Programs/miniconda3/envs/mne/bin/python
CPU                  arm (12 cores)
Memory               32.0 GB

Core
├☑ mne               1.4.0
├☑ numpy             1.23.5 (OpenBLAS 0.3.21 with 12 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.56.4
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.1
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.39.0 (OpenGL 4.1 Metal - 83.1 via Apple M2 Pro)
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.0.0
└☐ unavailable       ipympl

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2023-06-30:
Agreed `detrend=False` is better so DC is actually estimated, PR welcome!

chapochn, 2023-06-30:
ok, I'll give it a try.
It seems that same issue is present for the multitaper psd calculation, but I am not directly seeing where it is happening in the code.

chapochn, 2023-06-30:
inside _mt_spectra:
`x = x - np.mean(x, axis=-1, keepdims=True)`

Not sure why this is there?

larsoner, 2023-06-30:
It makes me think removal is intentional and required. Maybe we do need to remove it... I'd have to go back to some signals textbook to remember. It's possible that including DC when using a window function (which we do in `Welch` and use several in MT) is not a good idea.

chapochn, 2023-06-30:
I can see how there could be some leakage from the DC if ones pads the signal with 0s. But I believe it depends on the usecase. Sometimes one wants to know the DC. Sometimes not. Maybe would make more sense to have it as an option rather than set to removing the mean.

larsoner, 2023-06-30:
Yeah a `remove_dc=False` default kwarg makes sense!

PR #11769 (2023-08-23): added option to keep or remove the DC for psd calculations issue #11767
#### Reference issue
Fixes #11767

#### What does this implement/fix?
enhanced the functions `psd_array_welch` and `psd_array_multitaper` with option "remove_dc", with True as default, since the previous default behavior was to remove the DC. Updated the relevant documentation and all the related functions.
closed 2023-08-23 by drammock