mne-tools/mne-python #11768: make test not working locally

chapochn, 2023-06-30:
### Description of the problem

I followed all the instruction on https://mne.tools/stable/install/contributing.html. Here: https://mne.tools/stable/install/contributing.html#running-the-test-suite, I am not able to run `make test`

### Steps to reproduce

```Python
run "make test" from the mne-python root folder.
```

### Expected results

the tests would run

### Actual results

I get:

`make: *** No rule to make target `test'.  Stop.`

### Additional information

Platform             macOS-13.3.1-arm64-arm-64bit
Python               3.11.4 | packaged by conda-forge | (main, Jun 10 2023, 18:08:41) [Clang 15.0.7 ]
Executable           /Users/chapon01/Programs/miniconda3/envs/mnedev/bin/python
CPU                  arm (12 cores)
Memory               32.0 GB

Core
├☑ mne               1.5.0.dev45+gb92c88d70
├☑ numpy             1.24.4 (OpenBLAS 0.3.23 with 12 threads)
├☑ scipy             1.11.1
├☑ matplotlib        3.7.1 (backend=MacOSX)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.0
├☑ numba             0.57.1
├☑ nibabel           5.0.1
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.3
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.39.1 (OpenGL 4.1 Metal - 83.1 via Apple M2 Pro)
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.3
└☑ mne-qt-browser    0.5.1

Ecosystem (optional)
├☑ mne-bids          0.13.dev0
├☑ mne-connectivity  0.6.0dev0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline

larsoner, 2023-06-30:
I think we got rid of it at some point but forgot to remove it from the contributing doc. Really you can run `pytest` in a targeted way or `make flake` to run the style tests. We should probably recommend these two options. I wouldn't recommend a blanket `make test` because it will take 10-20 minutes to run locally, whereas if you `pytest mne/something/specific` it will be done in < 1 minute generally

chapochn, 2023-06-30:
ok thanks! I guess should update https://mne.tools/stable/install/contributing.html then...

larsoner, 2023-06-30:
Yeah I think that would be better than adding a generic `make test` target

PR #12042 (2023-10-01): DOC: Remove make test
Skipped CIs because it seems unnecessary based on `git grep "make test"` and `"make pytest"`

Closes #11768
Closes #10526
closed 2023-10-01 by drammock