mne-tools/mne-python #11772: BUG: raw.plot_psd(ax=ax) does not plot into ax

larsoner, 2023-07-03:
For me both of these are permitted but neither plots into the correct axes, both pop up a new figure:
```Python
>>> import mne
>>> import matplotlib.pyplot as plt
>>> raw = mne.io.read_raw_fif(mne.datasets.testing.data_path() / 'MEG' / 'sample_audvis_trunc_raw.fif').pick('eeg')
>>> fig, ax = plt.subplots()
>>> raw.plot_psd(axes=ax)
<MNELineFigure size 1000x350 with 2 Axes>
>>> raw.plot_psd(ax=ax)
<MNELineFigure size 1000x350 with 2 Axes>
>>> print(len(plt.get_fignums()))
3
```
Based on the [1.0 docs](https://mne.tools/1.0/generated/mne.io.Raw.html#mne.io.Raw.plot_psd) for example `ax=ax` should plot into the existing `ax` and `axes=...` should raise for invalid keyword.

@drammock can you look?

larsoner, 2023-07-04:
> For me, this one works correctly

Okay I checked on another system of mine and this case was okay there. So maybe I'm on some `dev` version of matplotlib on my other machine or something... I'll investigate more but hopefully we can ignore this for now!

> the keyword axes is not defined in plot_psd signature, I would not expect that one to work, but it does not raise.

Indeed this part at least seems to be a bug. It's especially weird since it *does* raise for other options like `invalid_kwarg`, but not `axes`.

drammock, 2023-07-04:
just a note that I can see why this is happening but haven't had a chance to implement the fix today, will do tomorrow.

PR #11778 (2023-07-11): Fix kwarg handling in legacy .plot_psd() method
closes #11772 

new behavior is

- any kwargs passed to legacy `raw.plot_psd()` that are not recognized params of that method but happen to be params of Spectrum.plot() now raise a warning. 
- those kwargs will be respected though, instead of overwritten with the fallback values drawn from the defaults of `plot_raw_psd()`.

@larsoner @mscheltienne LMK if you think this is a good approach... normally unrecognized kwargs will raise a TypeError (and we could do that here) but my feeling was that since this is a legacy method whose kwargs are getting [... cut]
closed 2023-07-11 by drammock