mne-tools/mne-python #11775: Allow Epochs reject and flat params to be callables

drammock, 2023-07-05:
The main use case is to do channel rejection based on standard deviations or absolute peak values (not *peak-to-peak* values, which is what we do currently)

This has come up a few times, e.g. here: 
https://mne.discourse.group/t/epoch-rejection-according-to-deviation-from-mean/2980

withmywoessner, 2023-07-21:
Hello @drammock! Is there anyway I could contribute to this? I am very much interested in this feature.

drammock, 2023-07-21:
sure. I've got it flagged as a potential project for an upcoming sprint, but I think we have enough such projects that there's no harm to remove this one.  Especially if you're motivated to do it soon!  LMK if you need any additional guidance, otherwise feel free to open a WIP-pull request and we can go from there.

withmywoessner, 2023-07-21:
Okay, thanks @drammock! I read the mne forum post linked and it mentions modifying reject and flat to accept functions. Is that the route I should go?

drammock, 2023-07-21:
yes. It would also be good to add an example to `examples/preprocessing` folder that shows how to reject epochs based on absolute peak values by defining and passing a suitable callable.