mne-tools/mne-python #11788: SNIRF reader can't handle file with MeasurementTime containing milliseconds

kdarti, 2023-07-09:
### Description of the problem

Can't read .snirf file with `/nirs(i)/metaDataTags/MeasurementTime` in the format `hh:mm:ss.sss`

### Steps to reproduce

```Python
import mne
raw_intensity = mne.io.read_raw_snirf(file, verbose=True).load_data()
```

### Link to data

https://file.io/qgHnp94ioNl7

### Expected results

File should be read without issues since milliseconds should be fine according to the snirf spec (file also passes snirf validator):

[image]

### Actual results

[image]

### Additional information

Platform             Windows-10-10.0.19044-SP0
Python               3.11.4 | packaged by conda-forge | (main, Jun 10 2023, 17:59:51) [MSC v.1935 64 bit (AMD64)]
Executable           C:\Users\kdahlslatt\Anaconda3\envs\mne\python.exe
CPU                  Intel64 Family 6 Model 158 Stepping 10, GenuineIntel (12 cores)
Memory               15.7 GB

Core
├☑ mne               1.4.2
├☑ numpy             1.24.3 (MKL 2022.1-Product with 6 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.57.0
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.2
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.39.1 (OpenGL 4.5.0 - Build 30.0.101.1404 via Intel(R) UHD Graphics 630)
├☑ pyvistaqt         0.0.0
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.5.1
└☐ unavailable       ipympl

Ecosystem (optional)
├☑ mne-nirs          0.5.0
└☐ unavailable       mne-bids, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2023-07-10:
@kdarti agreed this is a bug, do you want to try to fix it? I think these lines would need to be changed, probably just by adding another possibility to the list of attempted strings:

https://github.com/mne-tools/mne-python/blob/5608da261cc18f4813c1165baa8a6d5f64a1ea8c/mne/io/snirf/_snirf.py#L421-L434

Xiezhibin, 2023-07-13:
I'd be happy to try and fix the bug!

Xiezhibin, 2023-07-14:
Sure

The mne.io.read_raw_snirf() method cannot handle .snirf files with the nirs(i)/metaDataTags/MeasurementTime in the format hh:mm:ss.sss. However, ISO 8601 and [Python's datetime module](https://docs.python.org/3/library/datetime.html#datetime.time.utcoffset) support this format. The ISO 8601 time format options are as follows:

- HH:MM:SS.ffffff (if microsecond is not 0)
- HH:MM:SS (if microsecond is 0)
- HH:MM:SS.ffffff+HH:MM[:SS[.ffffff]] (if utcoffset() is not None)
- HH:MM:SS+HH:MM[:SS[.ffffff]] (if microsecond is 0 and utcoffset() is not None)
The microsecond value is a zero-padded decimal number with 6 digits.

I would also need the test data or any related resources to effectively validate and replicate the solution. If you could provide the data in a structured format or specify any specific details, it would be very helpful.

closed 2023-07-19 by larsoner

kdarti, 2023-08-15:
Sorry I didn't reply sooner, just got back from a long vacation. I just tried it on the file I used for the issue report (https://file.io/qgHnp94ioNl7) and it worked fine. Big thanks to @Xiezhibin for the fix!

PR #11804 (2023-07-19): ENH: add microsecond format support in read_raw_snirf
#### Reference issue
Example: Fixes #11788

#### What does this implement/fix?
This Pull Request implements the parsing of the MeasurementTime data in four different formats:

HH:MM:SS.ffffff (if microsecond is not 0)
HH:MM:SS (if microsecond is 0)
HH:MM:SS.ffffff+HH:MM[:SS[.ffffff]] (if utcoffset() is not None)
HH:MM:SS+HH:MM[:SS[.ffffff]] (if microsecond is 0 and utcoffset() is not None)