mne-tools/mne-python #11791: mne_logo.svg bottom text hard to read in dark mode

pllim, 2023-07-10:
### Description of the problem

https://mne.tools/stable/_static/mne_logo.svg is not really made for dark mode. Bottom text is hard to read.

### Steps to reproduce

```Python
Look at the rendered README file on GitHub when browser set to dark mode. I used Chrome but it should not matter which one.
```

### Expected results

Bottom text to be lighter color in dark mode.

### Actual results

Bottom text remains black, so very hard to read in dark background.

### Additional information

I happened to view the README here. I always have dark mode on.

drammock, 2023-07-14:
indeed it is!  Maybe we should replace with https://mne.tools/dev/_static/mne_logo_small.svg instead, which works equally well on light and dark...  I'm not sure if/how we could make an `.rST` readme show different images depending on browser (or GH?) light/dark settings. Do you know if this is even possible @pllim ?

pllim, 2023-07-14:
AFAIK there is no if-else in RST, so if you have one image that works in both, that would be great!

closed 2023-11-07 by larsoner

pllim, 2023-11-07:
Hmm still hard to read for me. 🤷 

[image]

larsoner, 2023-11-07:
Was closed slightly prematurely -- should be fixed tomorrow morning after CircleCI deploys and I open a quick PR to change it to use the `_gray.svg` variant

larsoner, 2023-11-08:
All better

[image]

pllim, 2023-11-08:
Indeed. Thanks!

PR #12176 (2023-11-07): Tweak logo for dark mode
Fixes #11791. I made the text a bit lighter (60% gray), which makes it easier to read in dark mode. It should be still quite readable in light mode as well.

Old (current) light:
[image]

Old (current) dark:
[image]

New light:
[image]

New dark:
[image]
PR #12184 (2023-11-08): Use gray logo (works in light and dark modes)
This should hopefully finally fix #11791.