mne-tools/mne-python #11794: ENH: HTML anchor support in reports

larsoner, 2023-07-11:
When linking to a report like:

- https://mne.tools/mne-bids-pipeline/1.4/examples/ds000117/sub-01_ses-meg_task-facerecognition_report.html

It would be nice if you could link to a specific section of the report using an anchor. Similarly, clicking on a button on the left should change the URL to reflect the anchor. In principle it seems like this should be doable using the section titles themselves, maybe with an enumeration (`-1`, etc.) if there are duplicates beyond the first one (since there is a `replace=False` option for duplicated titles).

hoechenberger, 2023-07-13:
We already have anchors and I regularly use them to link to specific sections, like so:

https://mne.tools/mne-bids-pipeline/1.4/examples/ERP_CORE/sub-average_ses-P3_task-P3_report.html#global-19

I right-click on the respective TOC entry, select the "open in new tab" action, and copy the generated URL.

But yes, we should update the URL displayed in the browser's location bar upon clicking on TOC entries. I believe there's setting in mkdocs-material for that.

larsoner, 2023-07-13:
> But yes, we should update the URL displayed in the browser's location bar upon clicking on TOC entries. I believe there's setting in mkdocs-material for that.

I think this shouldn't be affected by any `mkdocs-material` setting -- it should be part of `mne.Report` directly and thus separate from any mkdocs-related stuff. (Note this issue is for mne-python not mne-bids-pipeline...)

Also ideally as above we should make the anchors use the titles (with enumeration beyond the first one if there are dups) rather than `global19` which is not an informative anchor name...

hoechenberger, 2023-07-13:
Right, of course! I got a little confused there.

closed 2023-08-17 by larsoner