mne-tools/mne-python #11797: Resampling is confusing, so make `raw.decimate()` private

cbrnr, 2023-07-12:
Resampling to a lower frequency requires an anti-aliasing filter, which is what `raw.resample()` automatically does. However, there's also a `raw.decimate()` method, which does not include an anti-aliasing filter. This is very dangerous (even though it is of course documented) and very confusing to users. (In addition, `scipy.signal.decimate()` includes the anti-aliasing filter, so the terminology is also inconsistent.)

I suggest to make `raw.decimate()` private (e.g. renaming it to `raw._decimate()`), so that users only see `raw.resample()`, which is the only method they should ever use (and for those rare cases where you absolutely want to skip the anti-aliasing filter, they can still use the private method).

larsoner, 2023-07-12:
A number of reasons

1. Consistency with `epochs.decimate`
2. It can be much faster than `resample` if you get unlucky with the `rfft` radix/`next_fast_len` (watch out for primes :scream: !)
3. It's better not to use `resample` if you don't have to because some of its behaviors are less than ideal (e.g., assumes periodicity, see also #5136)
4. Removing it will break some people's currently valid workflows, and we shouldn't do that without very good reasons (I don't think this burden has been met)
5. Whatever other reasons we might have had to add `decimate` in the first place that I haven't listed here (feel free to look back at the history)

agramfort, 2023-07-12:
I respectfully disagree too. It’s much harder numerically to resample than low pass and decimate. It’s also faster. We should have a warning but please don’t deprecate decimate. I never use resample.

cbrnr, 2023-07-12:
Sure, although I don't really understand the reason. Resampling = low pass filter + decimation.

drammock, 2023-07-12:
@cbrnr it seems like making `decimate` private is a non-starter. Do you want to change this issue's title to reflect the smaller change proposed by @larsoner:

> We could add extra layers of protection other ways, though. Not sure if we do this already but we could check `info['lowpass']` and if's not some suitable amount below the Nyquist frequency (we have chosen such a level for `epochs.decimate(...)` already) we could by default raise an error by adding a new param like `on_bad_lowpass='error'` -- I don't love the name but hopefully you get the idea. Anyone who has correctly/safely done `raw.filter(..., h_freq=...).decimate(...)` will have no change in behavior, but users who are doing something dangerous will now get an error by default, which they can turn into a warning or ignore like usual with `on_*` kwargs.

or should we instead close as `not planned`?

cbrnr, 2023-07-13:
@agramfort our filter design as per EEGLAB is super great, but it has little to do with designing a filter for resampling. We're relying on SciPy for resampling, which includes appropriate anti-aliasing filters.

@drammock I've changed the title to reflect what might be a good compromise. We could add a new `antialiasing=True` parameter to `raw.resample()`, which if `False` would be the current `raw.decimate()` behavior. This would solve the problem of having two resampling-related methods directly in the object's namespace, which is the main problem I wanted to address.

agramfort, 2023-07-13:
> @drammock <https://github.com/drammock> I've changed the title to reflect
> what might be a good compromise. We could add a new antialiasing=True
> parameter to raw.resample(), which if False would be [...]

that would be ok for me.

> Message ID: ***@***.***>
>

larsoner, 2023-07-13:
> This would solve the problem of having two resampling-related methods directly in the object's namespace, which is the main problem I wanted to address.

Ahh I misunderstood then -- based on your top post I (mis?)read that the danger and risks of using `.decimate` was the primary issue at hand. My suggestion of a new `on_*` kwarg above indeed is meant to deal with this aspect and not the idea of unification/namespace deduplication.

>> We could add a new antialiasing=True parameter to raw.resample(), which if False would be the current raw.decimate() behavior. This would solve the problem of having two resampling-related methods directly in the object's namespace, which is the main problem I wanted to address.

> that would be ok for me.

Point (4) I brought up above is still relevant though:

> 4. Removing [the decimate method] will break some people's currently valid workflows, and we shouldn't do that without very good reasons (I don't think this burden has been met)

Our changing the API on people (especially when their code is fine!) has been a valid complaint of people for a while so I think our burden/justification for doing so should be fairly high. I still don't think the burden has been met to break people's code to achieve this naming clarification/unification.

Beyond that, I think the proposal to move only to `resample` is more drastic a change than it seems at first. We currently have `raw.decimate`, `raw.resample`, and `epochs.decimate`, `epochs.resample`, and `Epochs(..., decim=)`. Historically, `epochs.decimate` has been around since 0.10 and `raw.resample` and `Epochs(..., decim=...)` since before that I think -- so our naming/use of these terms has been established and used by people for over 7 years now. To truly fully unify these, we'd need to move for example `Epochs(..., resample=..., antialias=...)` and also remove `epochs.decimate`. To me anything short of this and we still have the same redundancy problem somewhere that we have to explain, but changing *all* of these (including `Epochs(..., decim=...)` will break a ton of end-user code.

One final conceptual point potentially in favor of keeping the functions separate -- the code and operations are actually quite different, and don't work in equivalent circumstances. Our `resample` relies on `scipy.signal.resample`, which 1) allows resampling to an arbitrary number of samples and 2) always (implicitly or explicitly) applies an antialiasing filter (there is no way to disable it). `decimate` on the other hand can only subsample by an integer factor (e.g., no 160 Hz to 100 Hz), so in the new API design people might think "oh I've low-passed so I'll set `antialias=False` and it should work" but it won't because they're trying to go from 250 Hz to 100 Hz for example.

cbrnr, 2023-07-13:
> we have a 2 version deprecation policy so we do this. Deprecate with messages and guarantee it works for the 2 next versions it it works now without deprecation warning. it's very standard in open source.

Yes, I absolutely agree! We could even make it longer, but we should not be afraid to break people's 5 year old code if we deem the change important enough.

> this assumes people read warnings and that people understand the code they were given. I agree it's annoying but the feedback we got many times suggests it's the reality.

100% with you. But this also means they don't read our docs, where we already highlight the dangers and differences of `decimate()`.

> can you see how explicit are our docstrings? Do we have see also sections? do we have the proper warnings? I would start there

Yes, we already have a lot of explanations, but maybe we can still improve them with what @larsoner has suggested. But I think at the end of the day, we won't be able to fully address the confusion between `decimate` and `resample` with only docs and warnings.

cbrnr, 2023-07-13:
Another thought, our `decimate` function is equal to slicing. So a name change to e.g. `slice` would probably also clear up some confusion.

drammock, 2023-07-13:
> Another thought, our `decimate` function is equal to slicing. So a name change to e.g. `slice` would probably also clear up some confusion.

Not equal to slicing. Slicing glass start, stop, step whereas decimate only has step. If it were called slice I'd be confused why there was only step

cbrnr, 2023-07-13:
> Not equal to slicing. Slicing has start, stop, step whereas decimate only has step. If it were called slice I'd be confused why there was only step

It also has a `start`, which is called `offset`. But yes, it always extends to the end of the signal, but for me the mental model of slicing is better than decimating (which literally means "taking every 10th sample", and even if you don't take it literally means downsampling with anti-aliasing).

cbrnr, 2023-07-18:
Yes, there are valid use cases for `decimate`, but this issue is not about removing this method.

I wanted to point out that it is very confusing to have two methods that do similar things. In particular, `resample()` should be what the vast majority of users are looking for, and `decimate()` is for special cases where users know what they are doing (mainly because it does not apply an anti-aliasing filter). In addition, `decimate` is an unfortunate name, because in signal processing it refers to changing the sampling rate _with_ an anti-alias filter.

Personally, I'd rename `decimate()` to something less ambiguous. I suggested `slice()`, but there were objections. I still think it's a good name even though there is no `end` parameter (but that could be added).

The other suggestions are not ideal, e.g. combining the two methods into one (difficult on a technical level) or improving the docs (doesn't solve the issue because people will need to have read the docs).

Happy to hear additional suggestions/ideas!

larsoner, 2023-07-18:
My vote is still for adding some `on_*` kwarg [as above](https://github.com/mne-tools/mne-python/issues/11797#issuecomment-1632639584) and keeping resample and decimate.

cbrnr, 2023-07-18:
Yes, sorry I forgot your suggestion @larsoner! This is actually my second preferred option if we can't/don't want to rename. I'm not sure how reliable `info["lowpass"]` is, but if we can rule out cases where this info is not accurate (meaning that e.g. there is no info at all), I'm OK with adding extra `on_*` args (and ideally setting it to `"raise"` by default). Is this the case? Because `resample()` is almost always the safer option, because an additional lowpass filter for good measure is better than no filter and then violating the Nyquist theorem.

larsoner, 2023-07-18:
> but if we can rule out cases where this info is not accurate (meaning that e.g. there is no info at all)

Typically in these cases we're conservative and just set/assume `info['lowpass'] = info['sfreq'] / 2.` so the `on_*` condition will be triggered. So in this sense it should be safe

closed 2023-08-02 by hoechenberger