mne-tools/mne-python #11800: IndexError raised when plotting ICA sources using evoked data

jCalderTravis, 2023-07-13:
### Description of the problem

Using ICA.plot_sources raises an IndexError under specific circumstances. The circumstances are:

-- User is plotting using evoked input
-- User requests to plot a selection of ICA components with gaps (e.g. [0, 2] does not work, but [0, 1, 2] does) 

### Steps to reproduce

```Python
import os
import mne
from mne.preprocessing import ICA

sample_dir = mne.datasets.sample.data_path()
sample_raw = os.path.join(sample_dir, 'MEG', 'sample', 'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_raw)

raw.crop(tmax=40)
raw.pick_types(meg='mag')
raw.load_data()

ica = ICA(n_components=10)
ica.fit(raw)

epochs = mne.make_fixed_length_epochs(raw)
evoked = epochs.average()

ica.plot_sources(epochs, picks=0)       # Works fine
ica.plot_sources(epochs, picks=[0, 1])  # Works fine
ica.plot_sources(epochs, picks=1)       # Works fine

ica.plot_sources(evoked, picks=0)       # Works fine
ica.plot_sources(evoked, picks=[0, 1])  # Works fine
ica.plot_sources(evoked, picks=1)       # IndexError
```

### Expected results

No index error and the corresponding plot.

### Actual results

Traceback (most recent call last):
  File "K:\Code\SR-switches-analysis-meg\scratch.py", line 25, in <module>
    ica.plot_sources(evoked, picks=1)       # IndexError
  File "C:\Users\Joshua\anaconda3\envs\mne2023\lib\site-packages\mne\preprocessing\ica.py", line 2226, in plot_sources
    return plot_ica_sources(self, inst=inst, picks=picks,
  File "C:\Users\Joshua\anaconda3\envs\mne2023\lib\site-packages\mne\viz\ica.py", line 112, in plot_ica_sources
    fig = _plot_ica_sources_evoked(
  File "C:\Users\Joshua\anaconda3\envs\mne2023\lib\site-packages\mne\viz\ica.py", line 689, in _plot_ica_sources_evoked
    color, style = label_props[ii]
IndexError: list index out of range

### Additional information

mne:              1.3.1
numpy:            1.23.5 {MKL 2022.1-Product with 4 threads}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=QtAgg}

sklearn:          1.2.1
numba:            0.56.4
nibabel:          5.0.1
nilearn:          0.10.0
dipy:             1.6.0
openmeeg:         2.5.5
cupy:             Not found
pandas:           1.5.3
pyvista:          0.38.3 {OpenGL 4.5.0 - Build 30.0.101.1692 via Intel(R) UHD Graphics 620}
pyvistaqt:        0.9.1
ipyvtklink:       0.2.2
vtk:              9.2.6
qtpy:             2.3.0 {PyQt5=5.15.6}
ipympl:           Not found
pyqtgraph:        0.13.2
pooch:            v1.7.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found

jCalderTravis, 2023-07-13:
If there is not a simple solution to this distinctive bug, I would be happy to look into the problem further myself.

larsoner, 2023-07-13:
This does seem like a bug, please do investigate @jCalderTravis !

PR #11808 (2023-07-19): Fix small bug that raised error under some conditions
Fixes #11800.

#### What does this implement/fix?
ICA.plot_sources raised an indexing error under specific conditions described in #11800. This error is no longer raised.

The problem was that ICA component numbers, as stored in the variable ``picks``, were being used to index into a variable ``label_props`` that describes the style in which lines should be plotted. ``label_props`` is as long as ``picks``, not as long as the total number of ica components. Hence, the index of the component in ``picks``, not the ICA component number itself,  [... cut]
closed 2023-07-19 by larsoner