mne-tools/mne-python #11802: MNE appends empty data at the end of Raw object after CNT importation

withmywoessner, 2023-07-13:
### Description of the problem

Hello! My lab has some old neuroscan cnt files that we are processing and I discovered a bug where MNE appends empty data to the end of the file after importation using `mne.io.read_raw_cnt()`

When I read the file in neuroscan the file is normal so I know it is not a problem with the file.

### Steps to reproduce

```Python
import mne
mne.io.read_raw_cnt('empty_end.cnt', data_format = 'int32').plot()
```

### Link to data

I cannot shorten the file because it removes the problem. 
Here is a link to the file:
https://wustl.box.com/s/nfcbstcegtbh954ixh8orzyteiz0wxk2

### Expected results

The expected results should be the data imports as normal and looks like this at the end:
[image]

### Actual results

Instead there is empty data at the end and when I run:
`mne.io.read_raw_cnt('/home/woess/current_assignment/empty_end.cnt', data_format = 'int32').to_data_frame().tail(100)`
The data looks like this:
[image]
The plot looks like this:
[image]

### Additional information

Platform             Linux-5.15.90.2-microsoft-standard-WSL2-x86_64-with-glibc2.31
Python               3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:08:06) [GCC 11.3.0]
Executable           /home/woess/mambaforge/envs/mne/bin/python
CPU                  x86_64 (12 cores)
Memory               11.3 GB

Core
├☑ mne               1.4.0
├☑ numpy             1.23.5 (OpenBLAS 0.3.21 with 12 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.56.4
├☑ nibabel           5.0.1
├☑ nilearn           0.10.0
├☑ dipy              1.6.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.3
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.38.5 (OpenGL 3.3 (Core Profile) Mesa 21.2.6 via D3D12 (AMD Radeon(TM) Graphics))
├☑ pyvistaqt         0.9.1
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.0 (PyQt5=5.15.6)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.2
└☑ mne-qt-browser    0.4.0

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

withmywoessner, 2023-07-13:
I think a solution to this problem would be to change this cnt.py file to stop reading after `n_samples` samples: [link to code](https://github.com/mne-tools/mne-python/blob/926bec285c34c0fbdcad7b09ac4f550af8288e8a/mne/io/cnt/cnt.py#L300C1-L303C18)
It is commented that this variable is not reliable but I am not sure why. Currently the file stops being read after `sample_stop` samples: [link](https://github.com/mne-tools/mne-python/blob/926bec285c34c0fbdcad7b09ac4f550af8288e8a/mne/io/cnt/cnt.py#L574C13-L575C87)
Where sample stop is equal to `sample_start + min(block_size // f_channels, data_left // f_channels - sample_start)`
I wrote a short script to return the `n_samples` variable myself and the number of samples it returned exactly matched where the end of the file was supposed to be, so maybe this problem can be fixed by:
`sample stop = sample_start + min((n_samples, block_size // f_channels, data_left // f_channels - sample_start))`

larsoner, 2023-07-13:
@withmywoessner want to try making this change and see if it breaks any tests?

We should probably add some kwarg to control whether the variable should be used or not... if someone wrote that it's unreliable then they probably have files where it's wrong.