mne-tools/mne-python #11810: No bounds checks for `{raw,epochs}.get_data()` with `tmin` or `tmax`

cbrnr, 2023-07-18:
I was a bit surprised to find out that `.get_data()` in combination with `tmin` or `tmax` arguments does not perform any bounds checks, but instead just returns the (entire) data array. I expected an error, a warning, or at least an info message stating that a given `tmin` or `tmax` argument exceeded the available data range.

Is this an intentional decision? If so, it should at least be documented. I think the current behavior is very dangerous, because it can lead to bugs that are really hard to find.

```python
n_channels = 32
sfreq = 256
data = np.ones((n_channels, 100 * sfreq))
info = mne.create_info(n_channels, sfreq)
raw = mne.io.RawArray(data, info)
onsets = np.arange(1 * sfreq, 91 * sfreq, sfreq, dtype=int)
events = np.column_stack(
    (onsets, np.zeros(len(onsets), dtype=int), np.ones(len(onsets), dtype=int))
)
epochs = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5)

epochs.get_data(tmin=-0.6, tmax=1.8)  # both tmin and tmax exceed data range!
```

drammock, 2023-07-18:
+1 to add an info message at least.  maybe a good candidate for `on_exceed_bounds='raise'|'warn'|'ignore'` ?

hoechenberger, 2023-07-19:
I would expect it to raise by default