mne-tools/mne-python #11816: How long do we keep "New in version ..." around?

cbrnr, 2023-07-20:
### Proposed documentation enhancement

I don't think we ever remove the "New in version ..." notices, but things like "New in version 0.9" seem a bit out of place nowadays. I suggest that we remove these notices once they are older than a certain amount of time (or certain amount of releases).

1. Are you OK with the general idea of removing old notices?
2. If so, which threshold would you suggest? We could remove them after 3 releases, similar to our deprecation warnings.

Note that since we make our documentation for previous versions available, it would still be possible to check when a specific method/attribute has been added (although this requires a bit more work then).

sappelhoff, 2023-07-20:
+1

1. yes
2. after 3 releases (not counting patch releases) seems reasonable to me

drammock, 2023-07-20:
what is the harm of leaving them in?

sappelhoff, 2023-07-20:
> what is the harm of leaving them in?

to me it's no "big harm" done in any way. However I think it does qualify as visual clutter when the information lies years in the past (and when I could still look it up by browsing older docs).

cbrnr, 2023-07-20:
Like @sappelhoff said, the documentation gets harder to read because of visual clutter. Take a look at various `mne.Epochs` methods, for example [`get_data()`](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.get_data).

In addition, I might ask the opposite: what's the benefit of leaving in very old messages like "New in version 0.9"?

drammock, 2023-07-20:
For another perspective: I often want to know what version of NumPy, SciPy or Matplotlib a function, method, or parameter was introduced (because I want to know if I can rely on it in MNE's codebase, given what our minimum supported versions are).  My experience has been that it can be really time consuming to figure it out.  I am assuming that downstream users of MNE might encounter the same situation and appreciate us leaving those notes in place indefinitely.

The basic problem is that the info isn't there for *every* function, method or parameter, and *we're not perfect about always adding those notes*.  So if there's *nothing* saying "new in version X" then a naive user **cannot safely assume** that it's been around forever, and they need to do some digging if they're worried about version compatibility.  By leaving them in, we save them some work and we save ourselves the trouble of removing them.

If the visual clutter is that bothersome, maybe we can explore ways to use CSS to make the notes less obtrusive?

cbrnr, 2023-07-20:
I don't think MNE is used as a library like NumPy, SciPy, or Matplotlib. People can also go and read our release notes, searching for the particular method should be reasonably fast. But in any case, visual clutter is indeed the main problem for me, even though now that you mentioned it, I don't like the inconsistency either.

drammock, 2023-07-20:
> I don't think MNE is used as a library like NumPy, SciPy, or Matplotlib.

what makes you say this?  Lots of folks on our forum have asked questions about embedding MNE plots in their own GUIs, for example.  There are also lots of libraries listed on [our "suite" page](https://mne.tools/dev/install/mne_tools_suite.html#related-software) that use MNE.

cbrnr, 2023-07-20:
Nowhere near NumPy etc. I don't have exact number of course, but my feeling is that the vast majority of people are using MNE as a package/app and not as a library. In any case, visual clutter is the main problem, as these notices are so prominent that they are basically the first thing you see.

drammock, 2023-07-20:
> Nowhere near NumPy etc

(?!) That doesn't mean we can just ignore that use case.

You keep saying "visual clutter is the problem" but you're not engaging with the (so far only) concrete suggestion to address it:

> maybe we can explore ways to use CSS to make the notes less obtrusive?

Would that be good enough, or do you insist that they must be *gone*?

mscheltienne, 2023-07-20:
I'm +1 to remove them when they get too old. I agree that it is useful to know when an argument was added for compatibility, but I do not consider those notes for version below 1.0 anymore. The API changed too much for me to support MNE version below 1.0 in my applications. 
I think 3 releases is a bit on the low-end, I would probably keep them in for 5 or 6 releases thus covering 2 to 3 years. Beyond that, I can't imagine someone trying to support an application using 2 versions that different.

larsoner, 2023-07-20:
I am with @drammock on this one -- I think the utility of knowing when something was added and convenience of having them in the docs outweighs the cost of a few extra lines / clutter in HTML.

Moreover, even though not everyone uses our lib the same way as numpy/scipy/matplotlib/sklearn, I think it's nice/helpful to follow their conventions and community practices as much as possible (see also the effort toward SPEC adoption). Thus as long as these libraries keep their `versionadded` around, it's a good reason for us to keep ours, too.

> maybe we can explore ways to use CSS to make the notes less obtrusive?

We could for example have the `Notes` section collapsible and collapsed by default. Same with `examples` and/or `See Also`.

> I agree that it is useful to know when an argument was added for compatibility

It's useful more more than this, too. Every month or so (at least!) I'll even look to see "was this available back when I was doing project X", look at the docstring, see versionadded, then check they PyPI date. This would be much slower for me without the `versionadded` (I know because @drammock said they are not always there!). There are probably other uses, too, but it's more than just thinking about library version compat.

cbrnr, 2023-07-21:
OK, let's focus on removing visual clutter then.

> We could for example have the Notes section collapsible and collapsed by default. Same with examples and/or See Also.

This doesn't solve the issue, because "New in version ..." notices are not contained in these sections. Look how prominent these are:

[image]

These shouldn't even have their own callouts, I suggest putting a small yellowish or even grayish label like this:

[image]

hoechenberger, 2023-07-21:
Oh that nice little label on the right is slick! I like that.

drammock, 2023-07-21:
yes I was imagining the same location and approximate size as you did @cbrnr. Looks like we're on the same page, I'll see if I can make it happen

PR #11821 (2023-07-24): shrink versionadded notes in docstrings
closes #11816 

First attempt worked pretty well, I thought:

[image]

Until I did a quick scan of `git grep versionadded` results, and checked some other API pages, and found this in `CSP.plot_filters`:

[image]