mne-tools/mne-python #11817: Full ERP CORE dataset not available?

cbrnr, 2023-07-20:
[ERP CORE](https://erpinfo.org/erp-core) consists of data from 40 participants and 6 different paradigms. MNE includes `mne.datasets.erp_core`, which consists of only one file (`ERP-CORE_Subject-001_Task-Flankers_eeg.fif`). Unless I am mistaken, there doesn't seem to be an option to either select a different subject, a different paradigm, or the entire ERP CORE dataset.

I think we should either document this, or even better, add support for the entire dataset (available [here](https://osf.io/thsqg/) as EEGLAB `.set/.fdt` files).

larsoner, 2023-07-20:
Okay with me, then we could use it in mne-bids-pipeline, too

cbrnr, 2023-07-21:
I can work on it in a couple of weeks, probably starting mid-September.

hoechenberger, 2023-07-21:
The idea here was to just provide a single file to play with (actually, to demonstrate metadata generation in the respective tutorial). 

The full dataset is available from the ERP core website.

I don't see much value in including it here, but I won't veto if somebody is willing to add it. I just don't know why and how it would be useful -- there's a BIDS version of the dataset and MNE-Python vanilla cannot handle this. So if anything, a dataset fetched should go into MNE-BIDS, but nobody would expect it there. Hence ... I'm not convinced. Again, I won't veto.

hoechenberger, 2023-07-21:
> Okay with me, then we could use it in mne-bids-pipeline, too

We already use it there. IIRC we fetch it directly from OSF.

larsoner, 2023-07-21:
> We already use it there. 

Yes I know we use the dataset :) By "use it" I mean "use the MNE fetcher" rather than our own custom code to download it.

hoechenberger, 2023-07-21:
Okay, I see

But again, this would require we download the BIDS version of the dataset, which MNE-Python vanilla cannot ingest. 

If at all, that fetcher should live in MNE-BIDS.

cbrnr, 2023-07-21:
I just thought it is confusing that the dataset is called ERP CORE when it is just one file. So yes, I like that proposal.

sappelhoff, 2023-07-21:
> If at all, that fetcher should live in MNE-BIDS.

let's not add a fetcher to mne-bids.

Let's rather go with your suggestion from here:

> Document the dataset is incomplete and from where the full version can be downloaded. Done.

**or**

make it required to have mne-bids installed when someone wants to download ERP CORE via the mne-python fetcher.

larsoner, 2023-07-21:
>> But again, this would require we download the BIDS version of the dataset, which MNE-Python vanilla cannot ingest.

> make it required to have mne-bids installed when someone wants to download ERP CORE via the mne-python fetcher.

FWIW we already have code that uses `read_raw_bids` with an MNE dataset (`tutorials/clinical/30_ecog.py`) so I don't consider this point a blocker for filling out the `erp_core` downloading functionality. Warning or erroring out when `mne-bids` isn't installed for such fetchers also seems like good idea if we do want to add them. But I'm also fine with just documenting better that we just have a minimal single-file downloader and that there are better/other ways to get the full dataset.

cbrnr, 2023-07-21:
Also, ERP CORE is also available as non-BIDS EEGLAB files.

cbrnr, 2023-08-17:
The documentation of this dataset already mentions that this is just a tiny subset of the original ERP CORE dataset. However, I think this is really hard to find, because it is only available here: https://mne.tools/stable/overview/datasets_index.html#erp-core-dataset

It would be nice to include this description in the docstring of `mne.datasets.erp_core.data_path()`, but this relies on [`_data_path_doc`](https://github.com/cbrnr/mne-python/blob/erp-core-dataset/mne/datasets/utils.py#L40), which currently does not have a field for additional notes. Are you OK with adding that, given that this section will be empty for all but one dataset (at least for now)?

larsoner, 2023-08-17:
Yeah that should be fine

cbrnr, 2023-08-18:
This turns out to be more work than anticipated. The entire docstring business is rather convoluted, and a nice solution will take me more time than I currently have. So unless someone else wants to do it, I'm fine with closing this issue and leaving everything as is (also because I seem to have been the first one to get confused about this dataset, maybe everyone else did find the relevant doc page).

closed 2023-08-24 by hoechenberger