mne-tools/mne-python #11818: Dc_offset in non-electrophysiology channels

scott-huberty, 2023-07-20:
This came up [in the discourse channel](https://mne.discourse.group/t/eyetracking-and-qt-browser-traces-of-x-y-position-disappear/7254) regarding Eyetracking channels, but could be relevant to any non electrophysiology channels.

The question being: should `remove_dc` only apply to electrophysiology channels? 

I think the general feeling is yes? If so, does `remove_dc` serve a double purpose of mean centering traces on the plot, and will removing it cause traces for some data to go beyond the current y-axis scalings?

Cc @mscheltienne @britta-wstnr @sappelhoff

larsoner, 2023-07-20:
I can see use cases for it. One probably easy way to get this is to have remove_dc start working like picks does elsewhere (can be "all", list of int, etc.) and leave False and True as convenience wrappers for what they do now (basically empty list and "data", respectively I think).

Unclear if we should change the default to "all", though. Probably safest to leave it as is.

mscheltienne, 2023-07-20:
My gut feeling is that we don't use the browser to inspect an exact Y-value but rather Y-range and shape. The scaling which defines the range is centered and thus the browser also expects centered data. Thus, DC-offset correction, or rather centering, seems fine.

mscheltienne, 2023-07-20:
Oh you are right that it's already available! I did not notice. Then +1 to change the behavior as proposed.

scott-huberty, 2023-07-21:
That's an interesting point @cbrnr ,  in theory I could see that causing weird behavior but I don't have a concrete example with any data, no!