mne-tools/mne-python #11824: Unnecessary duration annotation information imported from raw cnt files

withmywoessner, 2023-07-21:
### Description of the problem

When I read cnt files using `mne.io.read_raw_cnt()`, duration values are imported for annotations that are not needed. I don't think these values are necessary because Neuroscan already specifies the onset of the stimulus and response. The variable where these durations are acquired is latency. See neuroscan documentation below:
[image]
But the documentation I have seen online from [link](http://paulbourke.net/dataformats/eeg/) has not stated that this is used for anything.

### Steps to reproduce

```Python
mne.io.read_raw_cnt('test.cnt')
```

### Link to data

https://wustl.box.com/s/4ef9ay9nfqf58yoouo960c71o9ndh0qn

### Expected results

Plots should not have any durations in `cnt.annotations.durations` besides those of bad spans

### Actual results

There are durations in this field. They don't appear to convey any information of value:
[image]

### Additional information

Platform             Linux-5.15.90.2-microsoft-standard-WSL2-x86_64-with-glibc2.31
Python               3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:08:06) [GCC 11.3.0]
Executable           /home/woess/mambaforge/envs/mne/bin/python
CPU                  x86_64 (12 cores)
Memory               11.3 GB

Core
├☑ mne               1.4.0
├☑ numpy             1.23.5 (OpenBLAS 0.3.21 with 12 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=agg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.56.4
├☑ nibabel           5.0.1
├☑ nilearn           0.10.0
├☑ dipy              1.6.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.3
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.38.5 (OpenGL 3.3 (Core Profile) Mesa 21.2.6 via D3D12 (AMD Radeon(TM) Graphics))
├☑ pyvistaqt         0.9.1
├☑ ipyvtklink        0.2.2
├☑ vtk               9.2.6
├☑ qtpy              2.3.0 (PyQt5=5.15.6)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.2
└☑ mne-qt-browser    0.4.0

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

hoechenberger, 2023-07-22:
Are you suggesting these particular events should essentially be represented as annotation with a duration of zero (or one sample, if a zero duration is not possible)?

withmywoessner, 2023-07-22:
Yes, a duration of zero. Sorry if that wasn’t clear.

hoechenberger, 2023-07-22:
How would one differentiate between those instantaneous events and those which are actually supposed to have a duration? (I suspect those **do** exist too?)

withmywoessner, 2023-07-22:
@hoechenberger I am quite sure that the only events that are supposed to have a duration are 'bad blocks' (bad spans) in mne. These are specified with a separate parameter. But even those do not have a duration in neuroscan just a separate code for both the beginning and end of the 'bad block':
[image]
I already implemented the importation of bad blocks so they should be fine.

agramfort, 2023-07-23:
do you spot in
https://github.com/mne-tools/mne-python/blob/main/mne/io/cnt/cnt.py where
the pb could come from?

> Message ID: ***@***.***>
>

withmywoessner, 2023-07-23:
I think I can just replace these [lines](https://github.com/mne-tools/mne-python/blob/68e7272563fa88c2f0bd6d3e551eac9c51c2d07d/mne/io/cnt/cnt.py#L148C1-L149C73) with: 
```
duration = np.array(
            [0 for e in my_events], dtype=float
)
```

hoechenberger, 2023-07-23:
> duration = np.array(
>             [0 for e in my_events], dtype=float
> )

Probably `np.zeros_like()` could make that easier

withmywoessner, 2023-07-23:
> > duration = np.array(
> > [0 for e in my_events], dtype=float
> > )
> 
> perhaps `np.zeros_like()` could make that easier

Okay! I will do that! I just wanted to make as few changes as possible because it already looks like this:
https://github.com/mne-tools/mne-python/blob/68e7272563fa88c2f0bd6d3e551eac9c51c2d07d/mne/io/cnt/cnt.py#L148C1-L149C73

closed 2023-07-24 by larsoner