mne-tools/mne-python #11827: Test failures for MNE 1.4.2 in Fedora

penguinpee, 2023-07-22:
### Description of the problem

I'm busy upgrading MNE in Fedora to version 1.4.2. The package builds fine, but there are issues with some tests. In Fedora Rawhide Python 3.12 and Cython 3.0.0 have recently landed.

Tests just hung forever in our build environment. I'm currently running tests verbosely to see if it issues any warnings / errors. I also added `--timeout=180` to `pytest` in order to let the tests complete.

It was `mne/stats/tests/test_cluster_level.py::test_thresholds[NumPy]` test that just hung and needed the timeout for tests to proceed.

Here is the summary of test failures:

```python
XFAIL mne/viz/backends/tests/test_renderer.py::test_backend_environment_setup[foo]
ERROR mne/time_frequency/tests/test_spectrum.py::test_spectrum_errors - pytest.PytestUnraisableExceptionWarning: Exception ignored in: <function tq...
FAILED mne/decoding/tests/test_base.py::test_get_coef_inverse_transform[Scaler-kwargs0-True] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_inverse_transform[Scaler-kwargs0-False] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_inverse_transform[_Noop-kwargs1-True] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_inverse_transform[_Noop-kwargs1-False] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_multiclass_full[4-10-2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_multiclass_full[4-3-2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_multiclass_full[3-2-1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_get_coef_multiclass_full[3-1-2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_base.py::test_cross_val_multiscore - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_rank_deficiency - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_time_delaying_fast_calc[1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_time_delaying_fast_calc[cuda] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_receptive_field_1d[1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_receptive_field_1d[cuda] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_receptive_field_nd[1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_receptive_field_nd[cuda] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_receptive_field.py::test_inverse_coef - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_search_light.py::test_search_light - pytest.PytestUnraisableExceptionWarning: Exception ignored in: <function tq...
FAILED mne/decoding/tests/test_search_light.py::test_generalization_light - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_search_light.py::test_verbose_arg[1-True] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_search_light.py::test_verbose_arg[2-info] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/decoding/tests/test_search_light.py::test_cross_val_predict - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/io/bti/tests/test_bti.py::test_crop_append - pytest.PytestUnraisableExceptionWarning: Exception ignored in: <function tq...
FAILED mne/stats/tests/test_cluster_level.py::test_thresholds[NumPy] - Failed: Timeout >180.0s
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_large_n_samples[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_step_down_p[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_cluster_permutation_test[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_cluster_permutation_t_test[NumPy-ttest_1samp_no_p] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_cluster_permutation_t_test[NumPy-stat_fun1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_cluster_permutation_with_adjacency[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_cluster_signs[1samp-0.1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_cluster_signs[1samp-TFCE] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_cluster_signs[ind-0.1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_cluster_signs[ind-TFCE] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_adjacency_equiv[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_spatio_temporal_cluster_adjacency[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_permutation_test_H0[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_tfce_thresholds[NumPy] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-mask-shape0] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-mask-shape1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-mask-shape2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-indices-shape0] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-indices-shape1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[None-indices-shape2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[sparse-mask-shape0] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[sparse-mask-shape1] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
FAILED mne/stats/tests/test_cluster_level.py::test_output_equiv[sparse-mask-shape2] - DeprecationWarning: datetime.utcfromtimestamp() is deprecated and scheduled...
[... cut ...]
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.0
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ pandas            1.5.3
└☐ unavailable       numba, openmeeg, cupy

Visualization (optional)
├☑ vtk               9.2.6
└☐ unavailable       pyvista, pyvistaqt, ipyvtklink, qtpy, ipympl, pyqtgraph, mne-qt-browser

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

drammock, 2023-07-24:
thanks for this!  I'll try to take a stab at fixing it later this week unless someone beats me to it.  Fortunately at first glance these all look like pretty simple fixes (i.e., none of them are numeric failures, they're all dealing with deprecations or such)

larsoner, 2023-08-14:
https://download.copr.fedorainfracloud.org/results/gui1ty/neuro-sig/fedora-rawhide-x86_64/06203645-python-mne gives me a 404 now so I can't test. @penguinpee can you try the branch https://github.com/mne-tools/mne-python/pull/11878 to see if it works?

closed 2023-08-14 by larsoner

penguinpee, 2023-08-29:
@larsoner I was on vacation. I must have overlooked the notification on my return. But I see it has been merged meanwhile. So, I guess there's nothing left for me to do.

I saw that one of the co-maintainers of MNE in Fedora submitted a new issue (#11931) meanwhile regarding a test failure in 1.5.0.

PR #11878 (2023-08-14): MAINT: Work around NumPy deprecation
Closes #11827