mne-tools/mne-python #11840: Ability to filter certain channels but keep all channels.

Gvanderl, 2023-07-28:
### Describe the new feature or enhancement

Hello, 
This seems like a pretty fundamental feature so I feel like it might already be implemented. I couldn't find it in the documentation. 
When filtering the an mne.io.Raw object, it makes sense to use the "picks" parameter to apply a specific filter to a certain channel type only. However, doing so removes the other channels from the returned Raw and re combining the removed channels with the filtered ones seems like a pretty involved process. It would be simpler to just have a raw with certain channels filtered and others not. 

### Describe your proposed implementation

Add a "return_unfiltered_channels" to the filter method so that the returned object contains all channels when set to true. 

### Describe possible alternatives

Alternatively, there could be a function that combines the channels of raw objects so that we could have the filtered data on one side, and the leftover channels on the other, and combine the two after the fact. This is less elegant obviously.

agramfort, 2023-07-28:
Just use the picks parameter in https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.filter

Gvanderl, 2023-07-28:
Ah looks like I was using the wrong way to pick data. Thanks for the response

closed 2023-07-28 by Gvanderl