mne-tools/mne-python #11848: Problem in mne.preprocessing.ica._find_max_corrs

Wana999, 2023-08-01:
**Line 3178 in the ica.py file**
‘’‘
_mean_of_maps = np.std(np.asarray(maxmaps))_
‘’’

I think it should be
'''
_mean_of_maps = np.mean(np.asarray(maxmaps))_
'''

larsoner, 2023-08-01:
https://github.com/mne-tools/mne-python/blame/b5a9a25d3fac7e4058b91efcadf6561b85aa45a3/mne/preprocessing/ica.py#L3177-L3178

Yes that looks like a bug. I'm surprised it's working properly!

Would you be up for making a PR to fix it?

hoechenberger, 2023-08-02:
What are the implications of touching this? This piece of code is used for automated artifact detection, right? So if we change this, we may break existing z-score thresholds, yes? We should be very careful here. Even though it's a bug that needs to be fixed, I would treat this like a heavy API change that should not be introduced without a sufficiently long deprecation cycle (2 to 4 releases).

just my 2 ct

@agramfort WDYT?

Wana999, 2023-08-02:
I just discovered this question when I was learning how to analyze the correlation of the components of ICA. I don't really understand the algorithm flow of corrmap, so I hope this bug can be fixed as soon as possible.

hoechenberger, 2023-08-02:
It seems my assumption was incorrect and the EOG and ECG detection methods don't rely on `ICA.corrmap()`, so all should be good with fixing this asap.

hoechenberger, 2023-08-02:
You can try making a pull request with a fix if you want

larsoner, 2023-08-02:
The fix ended up being a littly bit bigger, PR in https://github.com/mne-tools/mne-python/pull/11858

larsoner, 2023-08-07:
Fixed by #11858, closing!

closed 2023-08-07 by larsoner