mne-tools/mne-python #11856: BUG: python setup.py creates mne.egg-info

larsoner, 2023-08-02:
`python setup.py sdist` creates a `mne.egg-info` in the root dir that can then mess up the `importlib.metadata` resolution of the package version. See discussion in https://github.com/mne-tools/mne-bids/pull/1160#discussion_r1281947981. The TL;DR is that @hoechenberger and I had versions reported as 1.4.2 when we were on 1.5.0.dev because this file sat around out of date.

Ideally this file would not be produced at all, but even `python -m build .` seems to create it.

cbrnr, 2023-08-16:
Where is this relevant? When doing `mne.__version__`? Can `importlib.metadata` be instructed to only use specific directories (not sure what exactly you are referring to though)?

larsoner, 2023-08-16:
The discussion linked above https://github.com/mne-tools/mne-bids/pull/1160#discussion_r1281947981 shows how/where it's relevant

> Can importlib.metadata be instructed to only use specific directories

Not that I know of but feel free to investigate

cbrnr, 2023-08-16:
It's probably not `pip`, because a minimal example is working as expected:

```
❯ tree
.
├── mypackage
│   └── __init__.py
└── pyproject.toml
```

```
❯ cat pyproject.toml           
[build-system]
requires = ["setuptools"]
build-backend = "setuptools.build_meta"

[project]
name = "mypackage"
version = "0.1.0"
```

```
❯ cat mypackage/__init__.py           
from importlib.metadata import version

__version__ = version(__package__)

print(__version__)
```

Install that package with

```
❯ pip install -e .
```

This creates the `mypackage.egg-info` folder. Test with

```
❯ python -c "import mypackage"
```

Now change the `version` to e.g. `"0.2.0"` in `pyproject.toml`, run `pip install -e .` again, and `python -c "import mypackage"` will show the updated version.

So I think the problem might be in the fallback `_version.py`?

sappelhoff, 2024-07-27:
@larsoner given the changes to our build system, can this be closed now?

larsoner, 2024-07-27:
Yes `setup.py` is gone and even just now doing `pip install --no-build-isolation -ve .` the problematic `mne.egg-info` was not created

closed 2024-07-27 by larsoner