mne-tools/mne-python #11861: Traceback when using spectrum.plot()

mscheltienne, 2023-08-04:
### Description of the problem

I do get the figure, but with the matplotlib traceback:

```
Traceback (most recent call last):

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/IPython/core/formatters.py:340 in __call__
    return printer(obj)

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/IPython/core/pylabtools.py:152 in print_figure
    fig.canvas.print_figure(bytes_io, **kw)

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/matplotlib/backends/backend_qtagg.py:75 in print_figure
    super().print_figure(*args, **kwargs)

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/matplotlib/backend_bases.py:2353 in print_figure
    restore_bbox = _tight_bbox.adjust_bbox(

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/matplotlib/_tight_bbox.py:28 in adjust_bbox
    ax.apply_aspect(locator(ax, None))

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/mpl_toolkits/axes_grid1/inset_locator.py:73 in __call__
    bbox = self.get_window_extent(renderer)

  File ~/pyvenv/mscheltienne/mne-python/lib/python3.10/site-packages/matplotlib/offsetbox.py:399 in get_window_extent
    renderer = self.figure._get_renderer()

AttributeError: 'NoneType' object has no attribute '_get_renderer'
```

### Steps to reproduce

```Python
from mne.datasets import sample
from mne.io import read_raw_fif

directory = sample.data_path() / "MEG" / "sample" 
raw = read_raw_fif(directory / "sample_audvis_raw.fif", preload=False)
raw.pick("meg").crop(0, 180).load_data()
raw.compute_psd().plot()
```
```

### Expected results

-

### Actual results

-

### Additional information

Version of MNE: `main`
Version of `matplotlib`: 3.7.2 with Qt5Agg backend

```
Platform             Linux-6.2.6-76060206-generic-x86_64-with-glibc2.35
Python               3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
Executable           /home/scheltie/pyvenv/mscheltienne/mne-python/bin/python
CPU                  x86_64 (12 cores)
Memory               31.0 GB

Core
├☑ mne               1.5.0.dev84+g0f2ea8183
├☑ numpy             1.23.5 (OpenBLAS 0.3.20 with 12 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.2 (backend=Qt5Agg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ numba             0.56.4
├☑ nibabel           5.1.0
├☑ nilearn           0.10.0
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.0.0
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.37.0 (OpenGL 4.6 (Core Profile) Mesa 22.3.5 via Mesa Intel(R) UHD Graphics 770 (ADL-S GT1))
├☑ pyvistaqt         0.10.0
├☑ ipyvtklink        0.2.3
├☑ vtk               9.2.6
├☑ qtpy              2.3.1 (PyQt5=5.15.2)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.3
└☑ mne-qt-browser    0.6.0.dev4+gc912752

Ecosystem (optional)
├☑ mne-bids          0.13.dev0
├☑ mne-connectivity  0.6.0dev0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline
```

mscheltienne, 2023-08-04:
It goes away if `raw.compute_psd().plot(average=True)` so it comes from something in https://github.com/mne-tools/mne-python/blob/193e41093e8f9c888c1d56abbc86267200c15e53/mne/viz/utils.py#L2586-L2631

mscheltienne, 2023-08-04:
And it seems to be limited to the Spyder console..

closed 2023-08-04 by mscheltienne

drammock, 2023-08-04:
probably duplicate of https://github.com/mne-tools/mne-python/issues/11799 which was reported upstream by https://github.com/matplotlib/matplotlib/issues/26287 and fixed by https://github.com/matplotlib/matplotlib/pull/26291