mne-tools/mne-python #11873: Checking the _find_outliers function

VladimirR46, 2023-08-10:
### Description of the problem

I seem to have found a bug in the _find_outliers function.
The essence of the method is iterative z-score evaluation, but on the second iteration we always have [nan nan ...] in the variable `this_z`. This is because the `zscore` function cannot work with the X variable after calling `X = np.ma.masked_array(X, my_mask)`
This mistake leads to the fact that we cannot find new outliers on the second iteration.

Maybe I just didn't understand something) Thanks everyone!

### Steps to reproduce

```Python
from mne.preprocessing.bads import _find_outliers

np.random.seed(33)
x = np.random.standard_normal(100)
x[0] = 9
x[1] = 4

idx = _find_outliers(x, max_iter=2, tail=0)
idx
```

### Expected results

[image]

### Actual results

[image]

[image]

### Additional information

Platform             Windows-10-10.0.22621-SP0
Python               3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
Executable           C:\Neurolab\Experiments\Children\ChildProcessing\Python\venv\Scripts\python.exe
CPU                  Intel64 Family 6 Model 154 Stepping 3, GenuineIntel (20 cores)
Memory               31.7 GB

Core
├☑ mne               1.4.2
├☑ numpy             1.24.3 (OpenBLAS 0.3.21 with 20 threads)
├☑ scipy             1.10.1
├☑ matplotlib        3.7.1 (backend=Qt5Agg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ pandas            2.0.1
└☐ unavailable       numba, nibabel, nilearn, dipy, openmeeg, cupy

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, ipyvtklink, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

drammock, 2023-08-10:
There is some weird edge behavior going on here. If I change

```py
x[0] = 9
x[1] = 4
```
into 
```py
x[3] = 9
x[11] = 4
```
then the result is correct.

drammock, 2023-08-10:
This might be a bug in SciPy:

```py
import numpy as np
from scipy.stats import zscore

rng = np.random.default_rng(seed=33)
x = rng.standard_normal(10)
mask = np.zeros_like(x)
y = np.ma.masked_array(x, mask)
y.mask[1] = True
print(zscore(y))
y.mask[1] = False
y.mask[0] = True
print(zscore(y))
```
I'll report upstream

VladimirR46, 2023-08-10:
Thank you, @drammock!

I repeated your code and got the same result. The problem must be in the `zscore` function.

drammock, 2023-08-10:
reported upstream: https://github.com/scipy/scipy/issues/19039

drammock, 2023-08-21:
upstream bug is fixed. Thanks again for reporting.

closed 2023-08-21 by drammock