mne-tools/mne-python #11874: add support to read continously sampled channel files from the Neuralynx acquisition system

KristijanArmeni, 2023-08-10:
### Describe the new feature or enhancement

After briefly discussing this with @drammock, opening this to  evaluate if there is interest in MNE-community to add support for reading the data from the Neuralynx acquisition systems (https://neuralynx.com/).

These electrophysiology time series are typically recorded from depth electrodes at high sampling rates (32KHz) to detect single-neuron spiking activity in humans. To my familiarity, a researcher typically will not interact with the raw dataset (due to large file sizes), but with a downsampled version of the continuous recording (`.ncs`) usually in combination with event files (`.nev`).

Information on the type of data files in a typical Neuralynx dataset:
- Neuralynx file specification: https://support.neuralynx.com/hc/en-us/articles/360040444811-TechTip-Neuralynx-Data-File-Formats
- FieldTrip docs: https://www.fieldtriptoolbox.org/getting_started/neuralynx/

Example dataset/research paper: 
- [A NWB-based dataset and processing pipeline of human single-neuron activity during a declarative memory task](https://doi.org/10.1038/s41597-020-0415-9) (organized following the [Neurodata Without Borders](https://www.nwb.org/) structure)

### Describe your proposed implementation

In brief, Neuralynx datasets are stored as a single `.ncs` (continuously sampled) files for each channel. A dataset directory containing only `.ncs` channel files could like something like this:

```bash
>>> ls ./neuralynx_dataset
LACC1.ncs   LAM5.ncs    LPCC1.ncs  LPHG5.ncs
LACC2.ncs   LAM6.ncs    LPCC2.ncs  LPHG6.ncs
LACC3.ncs   LAM7.ncs    LPCC3.ncs  LPHG7.ncs
LACC4.ncs   LAM8.ncs    LPCC4.ncs  LPHG8.ncs

```

It seems to me that on the user end this could look something like:
```python
raw = mne.io.read_raw_neuralynx("./neuralynx_dataset") # read all channel files and header information
```

or possibly for individual channel files:

```python
raw_ch1 = mne.io.read_raw_neuralynx_ncs("./neuralynx_dataset/LACC1.ncs") # read single channel file and header information
```

### Describe possible alternatives

The [Neo python package](https://neo.readthedocs.io/en/latest/index.html) has a [NeuralynxIO class](https://neo.readthedocs.io/en/latest/iolist.html#neo.io.NeuralynxIO) intended to be used as interface to read Neuralynx files in python, but offers no functionality to analyze and visualize the data.

### Additional context

I have not analyzed such data before, so I don't have a sample dataset at hand nor can give examples a typical workflow (but might in the coming months). I've only ever had to read in the header information of a Neuralynx dataset for purposes of documenting it and noticed this was not possible with MNE-Python.

larsoner, 2023-08-10:
> read all channel files and header information

To clarify this way of doing things -- I think it shouldn't *read all the data* (this should only be done when `preload=True`) but rather it should figure out how much data there are on disk, and read later. (This stuff is built into the BaseRaw class -- the `read_raw_neuralynx` "just" needs to know how many samples there actually are and how to read an arbitrary segment of data for a set of channel picks.)

> read single channel file and header information

I don't think we need to add this variant. If you want a single channel you would just use the directory version and pick as usual like:
```
read_raw_neuralynx(my_dir).pick(my_channel_names).load_data()
```
or so. And the data is only loaded from disk when calling `load_data` or `get_data` or using `preload=True` etc.

> The Neo python package has a NeuralynxIO class 

An optional dependency on `neo` seems okay to me. As long as that class does not immediately read all data and makes it easy to do on-demand reads and extract metadata, then we should be good to use it internally I think. Then the class becomes hopefully simple like the following (neo use is wrong here but just to give an idea):
```
class RawNeuralynx(BaseRaw):

    @verbose
    def __init__(fname, *, preload=False, verbose=None):
        try:
            import neo
        except Exception:
            raise ImportError(...) from None
        neo_class = neo.NeuralynxIO(...)
        last_samp = neo_class.n_samp  # or whatever the attribute is
        info = create_info('seeg', neo_class.sample_rate, neo_class.ch_names)
        super().__init__(...)

    def _read_segment_file(...):
        return neo_class.get_data(...)
```
It will be more complicated than this but hopefully this conveys the gist. 

A good template to follow would be https://github.com/mne-tools/mne-python/blob/main/mne/io/eximia/eximia.py -- it's easy to see the two methods that need to be implemented (`__init__.py` and `_read_segment_file`. Once those are implemented you get all the plotting/preload/picking etc. for free from the base class.

@KristijanArmeni any interest in giving it a shot?

KristijanArmeni, 2023-08-10:
Thanks @larsoner for the pointers and extra explanation! I'm happy to give this a shot (over the coming weeks). 

I think I've enough information to get started. Would be my first attempt at contribution, so will have to be baby steps, but will ping here in case I need further steering.

larsoner, 2023-08-10:
Awesome!

One other thing to think about is testing. We add files here (see instructions):

https://github.com/mne-tools/mne-testing-data

For this we'd want a file as small as possible from the system (e.g., ~1 sec and < 1 MB if possible) with known values, e.g., read using other software (like FieldTrip) and saved to `.mat`. Then at the MNE-Python end you can use your newly implemented reader and check the `raw.get_data()` values against the `read_mat(fieldtrip_mat)['data']` values using `assert_allclose`.

KristijanArmeni, 2023-09-06:
Just a small update on this. I started implementing `RawNeuralynx(BaseRaw)`. Need some pointers for next steps below.

### Summary 
Currently, `RawNeuralynx(BaseRaw)` uses [`NeuralynxIO()`](https://neo.readthedocs.io/en/latest/rawiolist.html#neo.rawio.NeuralynxRawIO) class from the [Neo package](https://neo.readthedocs.io/en/latest/index.html) to get basic file information from file headers, without loading data into memory (in my local install, neo required no additional dependencies on top of those included in mnedev). This seems to work ok (see screenshots in [Additional info](###Aditional-info) below)

### Next steps

To read data into memory, I need to make something like this work: `raw.pick(["LACC1", "LACC2"]).load_data()`

I need some help with:  
1. **Single channel files** For loading data into memory, I should implement `BaseRaw()._read_segment_file` method, but if I see correctly that one assumes that data on disk is stored in `shape = (channel, samples)` array? For Neuralynx, each file one disk is a single channel.
2. **Deprecation warnings**. I thought I would use `_read_segments_file` (following Eximia template) though after updating my local MNE copy, I’m getting deprecation warnings (`FutureWarning: mne.io.utils._read_segments_file is deprecated will be removed in 1.6, use documented public API instead. If no appropriate public API exists, please open an issue on GitHub.`). I get deprecation warning for `mne.io.meas_info.create_info` as well.
4. **Testing data**. I was not able to find a small dataset that could be used for testing (contacted Neuralynx if they have sample data, but no response). Not sure if I have any publicly shareable data at hand.

### Additional info

For example, for a neuralynx folder with channel files:
```bash
(base) kriarm@Kristijans-MacBook-Pro % ls /path/to/nlx_folder
LACC1.ncs	LAM1.ncs	LOINS1.ncs	LPHC1.ncs	LPreSMA1.ncs
LACC2.ncs	LAM2.ncs	LOINS2.ncs	LPHC2.ncs	LPreSMA2.ncs
LACC3.ncs	LAM3.ncs	LOINS3.ncs	LPHC3.ncs	LPreSMA3.ncs
LACC4.ncs	LAM4.ncs	LOINS4.ncs	LPHC4.ncs	LPreSMA4.ncs
LACC5.ncs	LAM5.ncs	LOINS5.ncs	LPHC5.ncs	LPreSMA5.ncs
LACC6.ncs	LAM6.ncs	LOINS6.ncs	LPHC6.ncs	LPreSMA6.ncs
LACC7.ncs	LAM7.ncs	LOINS7.ncs	LPHC7.ncs	LPreSMA7.ncs
LACC8.ncs	LAM8.ncs	LOINS8.ncs	LPHC8.ncs	LPreSMA8.ncs
LAINS1.ncs	LOFC1.ncs	LPCC1.ncs	LPHG1.ncs	LSMA1.ncs
LAINS2.ncs	LOFC2.ncs	LPCC2.ncs	LPHG2.ncs	LSMA2.ncs
LAINS3.ncs	LOFC3.ncs	LPCC3.ncs	LPHG3.ncs	LSMA3.ncs
LAINS4.ncs	LOFC4.ncs	LPCC4.ncs	LPHG4.ncs	LSMA4.ncs
LAINS5.ncs	LOFC5.ncs	LPCC5.ncs	LPHG5.ncs	LSMA5.ncs
LAINS6.ncs	LOFC6.ncs	LPCC6.ncs	LPHG6.ncs	LSMA6.ncs
LAINS7.ncs	LOFC7.ncs	LPCC7.ncs	LPHG7.ncs	LSMA7.ncs
LAINS8.ncs	LOFC8.ncs	LPCC8.ncs	LPHG8.ncs	LSMA8.ncs
```

The basic bookkeeping seems fine:
```python
In [6]: raw = RawNeuralynx(fname="/path/to/nlx_folder")
Inspecting files in /path/to/nlx_folder

In [7]: raw
Out[7]: <RawNeuralynx | LACC1.ncs, 80 x 2051175 (1025.6 s), ~65 kB, data not loaded>

In [8]: raw.info
Out[8]: 
<Info | 7 non-empty values
 bads: []
 ch_names: LACC1, LACC2, LACC3, LACC4, LACC5, LACC6, LACC7, LACC8, LAINS1, ...
 chs: 80 sEEG
 custom_ref_applied: False
 highpass: 0.0 Hz
 lowpass: 1000.0 Hz
 meas_date: unspecified
 nchan: 80
 projs: []
 sfreq: 2000.0 Hz
>

In [23]: raw._raw_extras
Out[23]: [{'orig_nchan': 80}]

In [24]: raw._filenames
Out[24]: 
['/path/to/nlx_folder/LACC1.ncs',
 '/path/to/nlx_folder/LACC2.ncs',
 '/path/to/nlx_folder/LACC3.ncs',
 '/path/to/nlx_folder/LACC4.ncs',
 '/path/to/nlx_folder/LACC5.ncs',
 '/path/to/nlx_folder/LACC6.ncs',
...
]
```

drammock, 2023-09-06:
> in my local install, neo required no additional dependencies on top of those included in mnedev

we already require `neo` for our doc building ([this example](https://mne.tools/dev/auto_examples/io/read_neo_format.html)) so it's not a huge deal to add it as a dependency for a neuralynx reader, IMO.

> I thought I would use `_read_segments_file` (following Eximia template) though after updating my local MNE copy, I’m getting deprecation warnings

are you getting that warning *within MNE's source code*? or are you testing it out in a separate script for now?  (Maybe time to open the WIP pull request so I could just look at the code and see what you're doing)

> I get deprecation warning for mne.io.meas_info.create_info as well.

This has moved, access it at `mne._fiff.meas_info.create_info` in package code (or at `mne.create_info` in user code)

> I was not able to find a small dataset that could be used for testing

maybe worth asking on the forum if anyone out there has a neuralynx system and would be willing to record us a short segment. Otherwise, maybe we could find a way to truncate and anonymize some of your non-sharable files?  Are you able to share them privately (e.g., to me and/or @larsoner) for that purpose, or are they 100% non-sharable?

larsoner, 2023-09-06:
> maybe worth asking on the forum if anyone out there has a neuralynx system 

Or if you have access to one, the data can come from a junk recording without the system hooked up to any subject/patient. As long as there are some non-zero non-NaN values in there we can work with it!

> Single channel files For loading data into memory, I should implement BaseRaw()._read_segment_file method, but if I see correctly that one assumes that data on disk is stored in shape = (channel, samples) array? 

Not exactly. The beauty of `_read_segment_file` is that you can read data from whatever file *or files (plural)* you need to read from to put the data in an array of shape `(n_channels, n_samples)`. So for your data format this would mean you probably need to open `n_channels` number of files instead of just one.

There are more details in there -- like you don't typically need to read all `n_channels` number of channels and there's a variable that tells you which to read -- but that's the general idea. If you can start by coding it to read all `(n_channels, n_samples)` worth of data within the `_read_segments_file` method and thensubselecting the correct set of data using something like `all_data[idx, start:stop]` then that's already a good start, and we can figure out how to make it efficient by only reading what needs to be read, etc.

> Maybe time to open the WIP pull request so I could just look at the code and see what you're doing

Agreed, our comments/feedback will probably be easier to make inline moving forward!

KristijanArmeni, 2023-09-07:
Great, thanks for the input, all makes sense! Good call -- I will create a WIP pull request and we can take it from there w.r.t. the issues raised above.

KristijanArmeni, 2023-09-19:
Update: we have a small Neuralynx test dataset!

It’s a small, 3.4 sec long (< 1MB) unconnected recording with 10 continuously sampled channels (`.ncs`), 10 spike files (`.nse`, likely empty though) and an event file (`.nev`). CLI print below.

In terms of moving forward:

1. Put the data to https://github.com/mne-tools/mne-testing-data?
1. Then continue the work in the draft PR https://github.com/mne-tools/mne-python/pull/11969?

### Additional info
Dataset contents:  
```bash
(base) kriarm@kristijans-mbp nlx % du -lh nlx_test
500K	nlx_test
(base) kriarm@kristijans-mbp nlx % ls -lh nlx_test
total 1000
-rw-rw-r--@ 1 kriarm  staff    17K Sep 19 12:21 Events.nev
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC1.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC2.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC3.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC4.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC5.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC6.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC7.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RACC8.ncs
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS1.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS2.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS3.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS4.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS5.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS6.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS7.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RACCuS8.nse
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RMCC1.ncs
-rw-rw-r--@ 1 kriarm  staff    30K Sep 19 12:21 RMCC2.ncs
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RMCCuS1.nse
-rw-rw-r--@ 1 kriarm  staff    16K Sep 19 12:21 RMCCuS2.nse
```

mscheltienne, 2023-09-19:
Great! Yes, the next step is to add the file to the testing dataset.

Antho2422, 2023-11-17:
@KristijanArmeni Hello ! Thank you for trying to add Neuralynx support in MNE. Any update on how this process is going ? I would be really interested !

Cheers, 
Anthony

larsoner, 2023-11-17:
Support was added in #11969, should be out in 1.6 which should be released soon