mne-tools/mne-python #11885: MAINT: Archive old doc versions

larsoner, 2023-08-15:
We currently have a version drop-down all the way to 0.11, which I think is rarely used or completely unused. Also when pushing commits to `mne-tools.github.io` repo I get warnings like this that suggest that our HTML has vulnerabilities, which is presumably due to ancient webpages:
```
remote: GitHub found 16 vulnerabilities on mne-tools/mne-tools.github.io's default branch (16 moderate). To find out more, visit:
remote:      https://github.com/mne-tools/mne-tools.github.io/security/dependabot
```
I suggest we put some lower bound on which versions to archive -- maybe the last 2 years worth of versions? This should help reduce this security issue, and also make the site faster to deploy and fetch, etc. If we don't want to get rid of the 0.11 docs entirely for example we could archive them in a `.zip` and link to them somewhere in our docs. This could easily be added as another step in our release process, and I don't think it would take too much extra time/effort (< 10 min) during release time.

Any objections? @drammock I'm releasing today but can take care of the changes this week and backport them to the `maint/1.5` branch I'm going to make (so that links to the archived versions exist).

drammock, 2023-08-15:
crossref: https://github.com/mne-tools/mne-python/issues/10932

I can't think of a good reason to keep the very old website versions available online. It would be nice if they existed *somewhere* --- could we have a separate branch of the `mne-tools.github.io` where we store *all versions* and then `main` can just have N versions?

> maybe the last 2 years worth of versions?

2 years would go back to 0.23, 3 years would include back to 0.21.

My feeling is that things are starting to stabilize and the package is maturing, which means people may become *less likely* to habitually update all the time, which means installs of a particular version might hang around for longer, so it might be good to preserve 3 years worth of docs instead of 2.

larsoner, 2023-08-15:
> It would be nice if they existed somewhere --- could we have a separate branch of the mne-tools.github.io where we store all versions and then main can just have N versions?

I was thinking we'd just create archives like `mne.tools/0.15.zip` and link to them in a bulleted list from somewhere in our docs. We could maybe access a new "Old docs" page from a `older...` dropdown entry in the `versions.json` that just takes you to the "Old docs" webpage with links to these zip archives. This seems very easy to maintain from our end but easy for people to download, unzip, and view if they really want to.

PR #11888 (2023-08-16): MAINT: Clean up whats_new and doc versions
- Closes #11885
- Closes #10932
- Closes #10789 

1. Refactor `whats_new` to show latest changes and link to all others
2. Refactor `rst_prolog` so we don't need to `.. include:: links.inc` and `.. currentmodule:: mne` all over the place
3. Add links to old doc version archives
4. Streamline `links.inc` to be much shorter and remove a lot of stuff we don't use (didn't check everything but checked some stuff at least)

@drammock let me know what you think. @sappelhoff also feel free to look since you had a preference for the long, continuous `whats_new [... cut]
closed 2023-08-16 by larsoner