mne-tools/mne-python #11892: Add interactive controls to `plot_evoked_topo`

wmvanvliet, 2023-08-17:
### Describe the new feature or enhancement

After a frustrating meeting with the professor and a PhD student where we wanted to take a look a the evoked responses from out pilot data, here's a request:

Could we please make `plot_evoked_topo` more interactive in the following ways:
 - Buttons / hotkeys for increasing/decreasing ylim
 - Buttons / hotkeys for selecting only mags, grads, joining grads
 - Legend present inside the new plots that open when you click on a channel

### Describe your proposed implementation

Added event handlers to the matplotlib figure generated by `plot_evoked_topo`.

### Describe possible alternatives

change and re-run your code very time you want to change any of the above.

larsoner, 2023-08-17:
Yes I think `plot_evoked_topomap` interactivity and options should also exist and be reused for `plot_topo`, and vice-versa (to the extent possible for the two plot types)

larsoner, 2023-08-17:
... or added to both (where possible) in the case that they're missing.