mne-tools/mne-python #11893: BUG: Doc drop-down can link to nonexistent HTML

larsoner, 2023-08-17:
Any `.html` that we have added in a given version that didn't exist in a previous version will errantly try to link to the nonexistent old page when using the drop-down (it assumes it'll be there). For example on the page:

- https://mne.tools/dev/changes/v0.20.html

Selecting "1.0" with the drop-down tries to go to a page that didn't exist back then (404):

- https://mne.tools/1.0/changes/v0.20.html

@drammock should we escalate to pydata-sphinx-theme?

drammock, 2023-08-17:
Already fixed in main of the theme. I will strive to find time to fix our remaining blocker and get a release out ASAP

larsoner, 2023-08-17:
Okay I'll close as an upstream bug that'll get automagically fixed soon :)

closed 2023-08-17 by larsoner